raspberrypi/pico-sdk #930: Unstable programming interface

martijnvwezel, 2022-07-18:
The pico RP2040 hardware dev board (not wifi).
The latest version [SDK 1.4.0 release](https://github.com/raspberrypi/pico-sdk/commit/2e6142b15b8a75c1227dd3edbe839193b2bf9041) 
The OS's it occurs: Windows and Ubuntu
Architectures: AMD64 and ARM64

Issue: Unstable programming interface

In the last month, a lot is the unstableness of updating the microcontroller, when connecting a new circuit board with the PICO programming/debugging is not an issue,
slower when writing the binary than it was months ago, but it works. After some time the pico does not want to respond anymore when you want to program it.
If you solder thick wires it is programable again, [the programmer I use](https://github.com/martijnvwezel/pico-probe-programmer) was always very stable and reliable.

When I check with my oscilloscope I see the clock line and data line communicating with the programmer. The weird thing is that it hangs it the error `DAP init failed`.
Making a thick connection with `GND` did not solve this issue.

The problem occurs with multiple different boards and different programmers.

JamesH65, 2022-07-18:
I presume you are talking about the SWD (Serial wire debug) interface? You should always ensure the shortest possible path from the host to the Pico when using SWD. Don't use any sort of breadboard in between, this will cause the problems you are describing. So, short wires, directly connected from, for example, the debugging Pi to the Pico.

This is covered in the Getting Started guide.

martijnvwezel, 2022-07-18:
Yes, I am talking about SWD, and of course shortest path possible. I did not experience any issues with communication before. Even lowering the clock did not help.

JamesH65, 2022-07-18:
So what has changed in your setup? The HW hasn't changed, and i don't think anything in the software stack that would cause this has changed.

martijnvwezel, 2022-07-19:
I guess windows, maybe a simple driver issue? I am not sure what the exact conditions are to trigger `DAP init failed`

lurch, 2022-07-19:
If you can reproduce this with a Raspberry Pi and two Raspberry Pi Picos, as detailed in Appendix A of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf I guess that would give a good "common starting point" for further investigation... but if you're using your own custom-built programming hardware, I guess the Raspberry Pi engineers can't really replicate that? :man_shrugging:

martijnvwezel, 2022-07-20:
@lurch  I want to understand what can trigger it, and the RPI engineers will possibly have more knowledge on this topic than I would have. Having to program hundreds of pico's the setup described in the getting started is not really an option.

JamesH65, 2022-07-20:
The trigger is usually too much resistance or capacitance in the cables which degrades the signal and makes communication erratic.

lurch, 2022-07-20:
Have you tried reducing the SWD speed with your current setup @martijnvwezel , to see if that makes things more reliable?

kilograham, 2022-07-20:
The title of this issue is very unclear.

Is the issue actually, that *your* version of a picoprobe appears to work less well when compiled against SDK1.4.0 vs SDK1.3.1?

martijnvwezel, 2022-07-20:
I reduced the speed of the SWD, but it did not help. I double-checked the GDB output log and there it stated it was set to 1KHz. Weird this is/was during measurement that the scope saw normal signals coming to the PICO. I am not sure how slock skew has an impact

weskerfoot, 2023-03-11:
I'm having a similar issue. I've connected the SWD pins on both picos using short wires and ensured the joints are good, yet I still get that "DAP init failed" error. I thought it might be due to udev rules, but I haven't had any luck with those either (rebooted several times as well). I also tried various usb ports and cables. I tried connecting a logic analyzer and the clock pin is held low while the SWDIO pin is held high, with no other activity (like it's not even trying to run). This is on an Arch Linux machine.

Running `openocd --debug -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000"` as root, I get these errors (see attached debug log)
[debug.log.txt](https://github.com/raspberrypi/pico-sdk/files/10949142/debug.log.txt)

kilograham, 2023-05-26:
closing as hasn't had an update in a while - this doesn't specify what you are using to connect to SWD; still better asked in the forums https://forums.raspberrypi.com/viewforum.php?f=143 or in the picoprobe repository