raspberrypi/pico-sdk #934: Unable to build ELF2UF2 when CMake Toolchain is not on PATH

t0rr3sp3dr0, 2022-07-20:
Environments like CLion bundle building tools and use them in CMake by setting the `CMAKE_MAKE_PROGRAM` variable, but this variable is not propagated to subprojects by default. ELF2UF2 is a subproject of the Pico SDK and it is not configured to propagate `CMAKE_MAKE_PROGRAM`. When the toolchain is not installed or not in the PATH, the build fails.

```text
====================[ Build | blink | Debug ]===================================
"/Users/pedro/Library/Application Support/JetBrains/Toolbox/apps/CLion/ch-0/221.5921.27/CLion.app/Contents/bin/cmake/mac/bin/cmake" --build /Users/pedro/repos/pico-examples/cmake-build-debug --target blink -j 3
[8/59] Performing configure step for 'ELF2UF2Build'
FAILED: blink/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure /Users/pedro/repos/pico-examples/cmake-build-debug/blink/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure 
cd /Users/pedro/repos/pico-examples/cmake-build-debug/elf2uf2 && "/Users/pedro/Library/Application Support/JetBrains/Toolbox/apps/CLion/ch-0/221.5921.27/CLion.app/Contents/bin/cmake/mac/bin/cmake" -GNinja /Users/pedro/repos/pico-examples/cmake-build-debug/_deps/pico_sdk-src/tools/elf2uf2 && "/Users/pedro/Library/Application Support/JetBrains/Toolbox/apps/CLion/ch-0/221.5921.27/CLion.app/Contents/bin/cmake/mac/bin/cmake" -E touch /Users/pedro/repos/pico-examples/cmake-build-debug/blink/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure
CMake Error: CMake was unable to find a build program corresponding to "Ninja".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "/Users/pedro/repos/pico-examples/cmake-build-debug/elf2uf2/CMakeFiles/CMakeOutput.log".
[9/59] Generating bs2_default_padded_checksummed.S
ninja: build stopped: subcommand failed.
```

This is what happens when you try to build the blink example (https://github.com/raspberrypi/pico-examples/tree/master/blink) with CLion. It uses Ninja as the CMake Generator, but Ninja is not installed on the computer and it uses the version shipped with the IDE. But when the build process reaches ELF2UF2, it fails because `CMAKE_MAKE_PROGRAM` is unset and Ninja is not in the PATH.

t0rr3sp3dr0, 2022-07-20:
Related:
- https://gitlab.kitware.com/cmake/cmake/-/issues/23355

kilograham, 2022-07-20:
Ha; thanks... i had never noticed this, even though I use CLion, because i have cmake and ninja installed locally too!

kilograham, 2022-10-21:
merged into develop

closed 2022-10-21 by kilograham

jtiscione, 2023-01-23:
I got a Raspberry Pi 4 to avoid troubles with setting up the toolchain, but that didn't do me any good; the instructions in the "getting started" tutorial are still broken. I can't get pico-examples to build in visual studio code.

[main] Building folder: pico-examples 
[build] Starting build
[proc] Executing command: /usr/bin/cmake --build /home/pi/pico/pico-examples/build --config Debug --target blink --
[build] [1/51   1% :: 0.041] Performing configure step for 'ELF2UF2Build'
[build] FAILED: blink/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure 
[build] cd /home/pi/pico/pico-examples/build/elf2uf2 && /usr/bin/cmake -GNinja /home/pi/pico/pico-sdk/tools/elf2uf2 && /usr/bin/cmake -E touch /home/pi/pico/pico-examples/build/blink/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure
[build] CMake Error: Error: generator : Ninja
[build] Does not match the generator used previously: Unix Makefiles
[build] Either remove the CMakeCache.txt file and CMakeFiles directory or choose a different binary directory.
[build] ninja: build stopped: subcommand failed.
[proc] The command: /usr/bin/cmake --build /home/pi/pico/pico-examples/build --config Debug --target blink -- exited with code: 1
[build] Build finished with exit code 1

kilograham, 2023-01-23:
try deleting your `/home/pi/pico-examples/build` directory and starting again.

jtiscione, 2023-01-25:
Yes that is what I missed.

I have a different problem now- the build completes successfully, but when I press the buttons for run or debug, it throws a segmentation fault.

jtiscione, 2023-01-28:
That was what I was doing wrong, I think.
But now what's stopping me is this message that appears whenever I try to run or run anything:
"Error: A serious error occurred with gdb, unable to continue or interrupt We may not be able to recover from this point. You can try continuing or ending session. Must address root cause though.

PR #935 (2022-10-16): Set CMAKE_MAKE_PROGRAM when building ELF2UF2 and PIOASM
Fix #934 by propagating `CMAKE_MAKE_PROGRAM` to ELF2UF2 and PIOASM.