raspberrypi/pico-sdk #943: Pi pico C/C++ SDK integration with VSCode

mic0331, 2022-07-25:
I am trying to get the Pico c/C++ SDK working with VS-Code on Mac OSX.

I set the SDK path in PICO_SDK_PATH and stored this path as an environment variable. Inside VSCode if I use the CMake Tool extension the build works correctly.

in the c_cpp_properties.json I explicitly included the SDK location

````json
"includePath": [
                "${workspaceFolder}/**",
                "/Users/MY_USER/pico/pico-sdk/**"
            ],
````

aallan, 2022-07-25:
Firstly, can you compile a project from the command line using CMake? It's probably best to make sure that you can get things working on the command line before looking at setting up VSCode.

Did you follow the instructions in [Getting Started with Raspberry Pi Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)? See Chapter 7, but also sections 9.1.2 and 9.1.3 in the Building on Apple macOS chapter.

mic0331, 2022-07-25:
Thanks, yes I can compile with the command line.
````
...
[ 50%] Building CXX object CMakeFiles/pioasm.dir/gen/parser.cpp.o
[ 60%] Building CXX object CMakeFiles/pioasm.dir/c_sdk_output.cpp.o
[ 70%] Building CXX object CMakeFiles/pioasm.dir/python_output.cpp.o
[ 80%] Building CXX object CMakeFiles/pioasm.dir/hex_output.cpp.o
[ 90%] Building CXX object CMakeFiles/pioasm.dir/ada_output.cpp.o
[100%] Linking CXX executable pioasm
[100%] Built target pioasm
[ 98%] No install step for 'PioasmBuild'
[100%] Completed 'PioasmBuild'
[100%] Built target PioasmBuild
````
I followed the instructions in the getting started guid.

lurch, 2022-07-26:
Not something I know much about, but perhaps some of the other settings from here are needed? https://github.com/raspberrypi/pico-project-generator/blob/master/pico_project.py#L1039
:man_shrugging:

jancumps, 2022-11-21:
For me this worked:
add a directory .vscode (if not there yet.
create a file settings.json, with this setting (or edit the existing one):
{
    "C_Cpp.default.configurationProvider": "ms-vscode.cmake-tools"
}

JamesH65, 2022-11-22:
That is pretty much what the pico-project-generator uses.

kilograham, 2023-02-09:
sounds like this is closable.

closed 2023-02-09 by kilograham

brianlmerritt, 2023-12-29:
The issues of squiggly lines are fixed in projects generated by pico-project-generator, but the above settings.json did not fix the issue on new Raspberry Pi OS, so I am adding here in case someone else stumbles on this.

In the file `c_cpp_properties.json` add the following content

```
{
    "configurations": [
      {
        "name": "Linux",
        "includePath": [
          "${workspaceFolder}/**",
          "${env:PICO_SDK_PATH}/**"
        ],
        "defines": [],
        "compilerPath": "/usr/bin/arm-none-eabi-gcc",
        "cStandard": "gnu17",
        "cppStandard": "gnu++14",
        "intelliSenseMode": "linux-gcc-arm",
        "configurationProvider" : "ms-vscode.cmake-tools"
      }
    ],
    "version": 4
  }
```