raspberrypi/pico-sdk #945: Software Interrupt SPI

0i9j, 2022-07-26:
I´m trying to get the Raspberry Pi Pico running in SPI Slave Mode without a ChipSelect / SlaveSelect Pin. 
My idea was, to handle the incoming data with an interrupt function.

Is it possible, to set up an interrupt function, that gets triggerd, when the RX register of my SPI port is full?

I have checked the SDK documentary and also tried to find examples on how to use SPI interrupts, but wasn´t able to find any. (except for CS/SS GPIO interrupts) As there is an Interrupt source for SPI interrupts in the SDK documentary, i assume i should be able to achive what i need.

In the following code the interrupt never gets triggerd. I have no idea why.

//setup of the SPI Pins on the Raspberry Board 
gpio_set_function(8, GPIO_FUNC_SPI); 
gpio_set_function(9, GPIO_FUNC_SPI); 
gpio_set_function(10, GPIO_FUNC_SPI); 
gpio_set_function(11, GPIO_FUNC_SPI);

//Setting Baudrate to 1000000 and spi1 to slave Mode 
spi_init(spi1, 1000 * 1000); 
spi_set_slave(spi1, 1);

//Setting Interrupt Source to SPI interrupt 
irq_set_enabled(SPI1_IRQ,1);

//setting up a function to handle the 
interrupt irq_set_exclusive_handler(SPI1_IRQ, spiIRQroutine);

//Interrupt Service Routine 
void spiIRQroutine() { 
           Serial.println("Interrupt"); //print "interrupt to console" 
           spi_write_read_blocking(spi1, 0, spi1in_buf, BUF_LEN); //read data on RX Pin to spi1in_buf

           for(int i=0; i< sizeof(spi1in_buf); i++) {

                      Serial.println(spi1in_buf); //write the RX buffer to console } }

//Blink program 
void loop() { 
           gpio_put(LED_BUILTIN, 1); 
           sleep_ms(4 * 1000); 
           gpio_put(LED_BUILTIN, 0); 
           sleep_ms(4 * 1000); }

lurch, 2022-07-26:
I see you're using `Serial.println` - are you using one of the Arduino environments for Pcio? They're provided by third-parties and not directly supported by Raspberry Pi. So you'll need to ask elsewhere for help with that.

If you're able to recreate your example using "just" the C-based `pico-sdk`, the Raspberry Pi engineers are more likely to be able to offer you help.

0i9j, 2022-07-27:
@lurch hi thanks. The pico-sdk library is included in my project so i am able to use all the commands. Is there a certain setting to be done to get started with the SPI Interrupt? To me it seems like I did all the settings i need to get an interrupt as soon as the PI´s RX register gets data.
My Settings:
irq_set_enabled(SPI1_IRQ,1);
interrupt irq_set_exclusive_handler(SPI1_IRQ, spiIRQroutine);

If you can´t help me, do you know where I could ask?

fivdi, 2022-07-27:
Take a look at how [picod](https://github.com/joan2937/picod) does it, it may help.
Part of the relevant code is [here](https://github.com/joan2937/picod/blob/master/DAEMON/picod.c#L1972-L1983).
Most likely `SPI_SSPIMSC_RXIM_BITS` needs to be set to enable interrupts.
The interrupt code posted above makes a lot of blocking calls. Normally, this isn't the best of ideas.

0i9j, 2022-07-27:
@fivdi Thank you! setting the SPI_SSPIMSC_RXIM_BITS did the job. For some reason i am still just able to read 8 bit data. 
I set the spi_set_format to 16 bit, and i am reading with the spi_read16_blocking into an uint16_t register. Still, when I wanna read the data on my Com Port i am just seeing the first 8 bit. do you know what i forgot to set?

fivdi, 2022-07-27:
> do you know what i forgot to set?

I afraid I don't.

lurch, 2022-07-28:
"when I wanna read the data on my Com Port" - are you sure that whatever "Com Port" code you're using is able to display 16-bit values, and isn't just truncating the result to 8 bits?

closed 2022-09-07 by 0i9j