raspberrypi/pico-sdk #950: compile_time_choice.S fails to build when using PICO_CONFIG_RTOS_ADAPTER_HEADER

NomiChirps, 2022-07-28:
The PICO_CONFIG_RTOS_ADAPTER_HEADER define, introduced in https://github.com/raspberrypi/pico-sdk/pull/851, will cause [src/common/pico_base/include/pico/config.h](https://github.com/raspberrypi/pico-sdk/blob/master/src/common/pico_base/include/pico/config.h) to include the named file. When that pico/config.h is included from [src/common/pico_base/include/pico.h](https://github.com/raspberrypi/pico-sdk/blob/master/src/common/pico_base/include/pico.h), this works fine.

However, [boot_stage2/compile_time_choice.S](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/boot_stage2/compile_time_choice.S) includes [boot_stage2/include/config.h](https://github.com/raspberrypi/pico-sdk/tree/master/src/rp2_common/boot_stage2/include/boot_stage2/config.h), which includes pico/config.h DIRECTLY. This means that the __PICO_XSTRING() macros aren't defined and preprocessing fails.

I ran into this issue while using https://github.com/dfr/rules_pico to build the SDK, but from my reading of the CMake files here, I don't see how it would work for a regular build either.

kilograham, 2022-07-28:
boot_stage2 build in the SDK does not link against FreeRTOS-Kernel and so would not include this define (so i'd say to that extent this is a bug in rules_pico)

There is a bug to fix that boot_stage2 should include the full pico config chain now which is assembly safe

NomiChirps, 2022-07-28:
Yeah, that's fair. In rules_pico that define is associated with pico/config.h and gets propagated transitively to everything that depends on it, rather than being associated with the FreeRTOS library. I think it's still worth supporting this for consistency's sake, so that different compilation units aren't using different versions of the config (even if it makes no difference in this case).

Thanks for accepting this issue!

dfr, 2022-08-08:
Thanks for taking a look at this. Let me know if this is better addressed in rules_pico rather than in the SDK.

mordae, 2023-01-07:
I am confused by the above discussion.

I have decided to hook into these macros for compatibility with a simple home-grown task scheduler and the build fails with:

```
pico-sdk/src/common/pico_base/include/pico/config.h:23:10: error: #include expects "FILENAME" or <FILENAME>
   23 | #include __PICO_XSTRING(PICO_CONFIG_RTOS_ADAPTER_HEADER)
      |          ^~~~~~~~~~~~~~
```

Build works fine when I modify the `pico/config.h` header to include `pico.h` first. I `ifndef __ASSEMBLER__` inside the extra header.

I feel this needs a little bit better user story but I have not dived deep enough into the SDK to be able to propose a solution.

kilograham, 2023-01-25:
fix merged into develop

PR #1190 (2023-01-24): Remove (legacy) direct inclusion of 'pico/platform.h' and 'pico/config.h'
Now `pico.h` is safe to include from everywhere (including assembly files), and ensures that defines and configs are created in the right order

fixes #1170
fixes #950
closed 2023-01-25 by kilograham