raspberrypi/pico-sdk #951: vcocalc.py script should consider PLL configurations with REFDIV > 1

alastairpatrick, 2022-08-01:
This is a feature request. You can assign it to me. I have already written the code and will follow this up with a pull request if there are no objections.

At present, the vcocalc.py script only considers PLL configurations with REFDIV = 1. Sometimes it is beneficial to use PLL configurations with REFDIV > 1. For example, in an audio application using the common 44.1kHz sample rate, one might desire to use a system clock of 88.2MHz or 176.4MHz. At present, vcocalc.py cannot satisfy that exactly:
```
> python vcocalc.py 88.2
Requested: 88.2 MHz
Achieved: 88.28571428571429 MHz
FBDIV: 103 (VCO = 1236 MHz)
PD1: 7
PD2: 2
```
In this case, it would help if it considered REFDIV = 2, which yields an exact match:
```
> python vcocalc.py 88.2
Requested: 88.2 MHz
Achieved: 88.2 MHz
REFDIV: 2
FBDIV: 147 (VCO = 882.0 MHz)
PD1: 5
PD2: 2
```
As an aside, if this change is made, it might be worth adding a variant of set_sys_clock_pll() taking a ref_div parameter.

lurch, 2022-08-01:
ping @liamfraser

lurch, 2023-02-23:
This can be closed, as it's included in SDK 1.5.0 ?

kilograham, 2023-05-26:
fixed in 1.5.0

closed 2023-05-26 by kilograham