raspberrypi/pico-sdk #963: Assert error in types.h

canique, 2022-08-08:
The method
`static inline void update_us_since_boot(absolute_time_t *t, uint64_t us_since_boot)`
has an assert statement:
assert(us_since_boot <= INT64_MAX);

In a project the assert would fail while the debugger claims that us_since_boot is less than 1 million.

GCC: 9.2.1 arm-none-eabi
Project built in debug mode
Debugging with a 2nd Pico

JamesH65, 2022-08-08:
Should that be UINT64_MAX ?

canique, 2022-08-08:
I'm not sure.

But even if it was INT64_MAX, the assert should pass. Even with an INT32_MAX it should pass, given the fact that "us_since_boot" is so small in my example.

canique, 2022-08-08:
It's strange - I cannot reproduce the problem with the debugger anymore.
But: when I keep the original SDK code, my alarm does not trigger now.

When I change the INT64_MAX to UINT64_MAX, it's working.

kilograham, 2022-08-08:
> Should that be UINT64_MAX ?

nope. it is INT64_MAX so time deltas can be negative or positive

kilograham, 2022-08-08:
try the `develop` branch of this repo

kilograham, 2022-08-08:
can you please provide a repro if there is an issue

canique, 2022-08-14:
> try the `develop` branch of this repo

I checked out the develop branch today (without applying any custom changes to it) and the problem with the alarm not firing has disappeared.
So, whatever the reason was, it's been fixed in the develop branch. Thank you.

closed 2022-08-14 by canique

lurch, 2022-08-14:
Possibly fixed by #936 ?