raspberrypi/pico-sdk #966: uninitialized_ram vs uninitialized_data

Gadgetoid, 2022-08-10:
I've been digging into skipping zero-fill initialization for large bss sections and came across the `uninitialized_data` section in `memmap_default.ld`:

https://github.com/raspberrypi/pico-sdk/blob/2e6142b15b8a75c1227dd3edbe839193b2bf9041/src/rp2_common/pico_standard_link/memmap_default.ld#L177-L180

However platform.h appears to use `uninitialized_ram` which is not present in the linker script(s):

https://github.com/raspberrypi/pico-sdk/blob/2e6142b15b8a75c1227dd3edbe839193b2bf9041/src/rp2_common/pico_platform/include/pico/platform.h#L163

Are these intended to be one and the same?

If not, what is `uninitialized_data` intended for? I can't find a single use of nor reference to it outside of its definition in the linker script.

kilograham, 2022-08-10:
Hmm. This is really weird. yes they are supposed to be the same, but for some reason that section is included in the binary anyway, and even more strangely, *in exactly the right place*

kilograham, 2023-01-25:
fix is in #1111

kilograham, 2023-01-27:
merged into develop

closed 2023-01-27 by kilograham