raspberrypi/pico-sdk #967: How to build when I can't add `pico_sdk_init()` to existing project?

parbo, 2022-08-10:
Normally, you'd just specify a `CMAKE_TOOLCHAIN_FILE` when invoking `cmake`. Why does the Pico SDK require modifying the `CMakeLists.txt`? Is there another way?

lurch, 2022-08-11:
I'm not entirely sure what you're asking, but perhaps some of the comments in https://github.com/raspberrypi/pico-feedback/issues/210 will point you in the right direction? (But if they don't, please excuse my ignorance!)

parbo, 2022-08-12:
Yeah, that's the same problem. For my use-case I can solve it like that. But in general it'd be nice to be able to build third party libraries without modifying them or adding them as submodules to a parent project that pulls in the pico stuff. Feel free to close this if that's not something you want to do.

kilograham, 2022-08-12:
One issue with using `CMAKE_TOOLCHAIN_FILE` is that the SDK build also builds native tools used by the build (which was deemed easier for the majority of users/platforms). We may revisit this in the future.

Note also that today, libraries that depend on the SDK tend to be INTERFACE libraries (again we may make it easier to build static libraries dependent on the SDK in the future). if however you are building a pre-existing non Pico specifc static library (without pulling it into your project as a submodule), then surely you can just build it. why do you need to do `pico_sdk_init()` at all?

ooxi, 2022-08-17:
@parbo I replaced `project` with a macro to work around this

ricardomatias, 2023-05-21:
@ooxi could you elaborate?

kilograham, 2023-05-26:
seems like there is a workaround here

closed 2023-05-26 by kilograham