raspberrypi/pico-sdk #972: README proper git clone pico-sdk in Getting started

eriklundh, 2022-08-16:
I sent someone to use pico-sdk, they had some trouble, and I checked the README.
The instruction to download pico-sdk in teh README section "Quick-start your own project" reads 
"a. git clone this Raspberry Pi Pico SDK repository"
The current "Getting started with Raspberry Pi Pico" says on page 6
$ git clone -b master https://github.com/raspberrypi/pico-sdk.git
$ cd pico-sdk
$ git submodule update --init
The latter works with all the submodules and the sdk downloaded according to the datasheet has a much smaller footprint than eg git clone --recursive https://github.com/raspberrypi/pico-sdk.git
I suggest that the README get aligned with this.
I have not delved into if the other metods with automatic git clone through Cmake directives work correctly with submodules, but the README for those alternative install methods also might need some tender care.

ooxi, 2022-08-17:
I have trouble understanding the change you suggest. Could you please elaborate?

kilograham, 2023-05-26:
not sure i understand the issue; cloning submodules is not mentioned in the README.md. as mentioned #772 avoids recursive submodule init from `pico_sdk_import.cmake`

closed 2023-05-26 by kilograham

eriklundh, 2023-05-26:
Sorry that I missed your request for more information. At the time, last year, I was working on a series of boards to augment and promote the use of Pico in thousands of niche applications. I was concerned that seasoned engineers (and students) that we had planned to introduce to Pico-based short series product development at scale (in 60 countries) - would get confused by the procedure in the README. But I had to cancel the project since Pi Ltd stonewalled it at a critical juncture, luckily just before we started mass production of our boards. I thus no longer have any personal interest in the ease-of-use or success of the Pico ecosystem or Pi Ltd.

P33M, 2023-05-26:
Oh dear! Clearly we dropped the ball here. Can you provide some examples of these "niche applications"? No need to provide thousands of examples, a dozen will do. Also, we would be very interested to know your expected numbers of shipped units per annum.

I find it curious your claim that "Pi Ltd" (which I take to mean Raspberry Pi Ltd) stonewalled potential a customer. When did this happen? Did we ignore a purchase order, or any other request for parts?

eriklundh, 2023-05-27:
P33M, I think I understand your role. The products were designed to help PI resellers convert single board customers to small to medium volume customers that still bought the parts from the reseller, by giving application creators a much easier path to create a simple base pcb to connect modules instead of a nest of cables - “nest of cables”-built applications are hard and costly to duplicate - plus I also designed a way to apply CM4 with the same ease as Pico on any application PCB. The goal was to make it much easier for resellers to attract an large audience of engineers to build thousands of niche applications - a long tail of thousands applications amounts to a huge volume together - with an easier path based on surface mount techniques “on the kitchen table” in small numbers, without redesign at any manufacturing service in larger numbers. The target market was a resellers blue ocean of customer applications where there was a need for more than single quantity, but where the applications did not attract any “sharks” from a resellers perspective (which would turn the ocean red), i e small or large consultants that often recommend redesign with other cheaper parts - billable hours are their only product  and consultant often try offset their cost by recommending low cost manufacturing far far away - and distributors that are keen to take over the resellers customer once there are any potential for volume.

If you really want to pursue this, please contact me directly. I appear with my real name here. 
But I am done “keeping up appearances”