raspberrypi/pico-sdk #975: hardware_adc missing dependency on hardware_gpio

ooxi, 2022-08-17:
When including [`hardware/adc.h`](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_adc/include/hardware/adc.h) while depending on `hardware_adc_headers` I'm getting a compile error because [`hardware/gpio.h`](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_gpio/include/hardware/gpio.h) cannot be found

```
In file included from /mysource.c++:4:
/pico-sdk/src/rp2_common/hardware_adc/include/hardware/adc.h:12:10: fatal error: hardware/gpio.h: No such file or directory
   12 | #include "hardware/gpio.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```

This is caused by neither `hardware_adc` nor `hardware_adc_headers` declaring a dependency on `hardware_gpio` / `hardware_gpio_headers`.

kilograham, 2022-09-12:
all dependencies issues are moving to 1.5.0

kilograham, 2023-01-29:
fixed by #1211

kilograham, 2023-01-30:
merged into develop

closed 2023-01-30 by kilograham