raspberrypi/pico-sdk #979: CYW43 support only if PICO_BOARD=pico_w

Memotech-Bill, 2022-08-18:
The various CMake link library options **pico_cyw43_arch_...** only have any affect if the CMake call defines **PICO_BOARD=pico_w**. Otherwise they default to no-op.

This causes problems if (for example) the Pico W is plugged into a VGA Demo Board. Creating a **vgaboard_w.h** file, which is a copy of **vgaboard.h**, but with the **#include "boards/pico.h"** replaced by **#include "boards/pico_w.h"** does not resolve the problem.

The simplest work around I have found is to include the following very early in the CMake file:

    set(PICO_CYW43_SUPPORTED 1)

I have yet to work out whether this needs to be above **pico_sdk_init()** or above **include(pico_sdk_import.cmake)**.

erdometo, 2023-06-05:
If you are building from command line, no problem. But VSCode uses its own cmake extension to lint errors. That extension has no setting to pass extra arguments. Only fix is set(PICO_CYW43_SUPPORTED 1) in cmakelists.txt.

kilograham, 2024-08-09:
We plan to add a `PICO_CARRIER_BOARD` or similar in the future

lurch, 2025-07-02:
This is also (briefly) mentioned in https://datasheets.raspberrypi.com/rm2/rm2-datasheet.pdf 🙂