raspberrypi/pico-sdk #980: cyw43_arch_threadsafe_background bugs preventing deinit() / init()

Memotech-Bill, 2022-08-19:
I have code which is linked using the CMake cyw43_arch_threadsafe_background link_library.
The code can call cyw43_arch_init(), cyw43_arch_deinit() and later cyw43_arch_init() again.

The second call to cyw43_arch_init() hangs.

I have found at least two issues which are responsible for this:

* In cyw43_arch_threadsafe_background.c routine cyw43_arch_init() includes the line:

`    gpio_add_raw_irq_handler_with_order_priority(IO_IRQ_BANK0, gpio_irq_handler, CYW43_GPIO_IRQ_HANDLER_PRIORITY);`

  However cyw43_arch_deinit() is missing the corresponding:

`    gpio_remove_raw_irq_handler (IO_IRQ_BANK0, gpio_irq_handler);`

* Secondly, in file cyw43_arch_poll.c, the call to lwip_init() is wrapped in code to ensure that it is called only on the first call to cyw43_arch_init():

```
    static bool done_lwip_init;
    if (!done_lwip_init) {
        lwip_init();
        done_lwip_init = true;
    }
```

  However, the call to lwip_init() in cyw43_arch_threadsafe_background.c is missing a similar wrapper.

lurch, 2022-08-19:
See also #968 and #969 ?

kilograham, 2023-01-18:
should be fixed by #1177; please give it a try if you can - i haven't yet tested this particular issue

kilograham, 2023-01-24:
#1177 merged into develop

closed 2023-01-24 by kilograham

flaviorenga, 2026-02-15:
Hi @Memotech-Bill and all!
I'm working on a Pico 2W project that needs to power cycle the BLE module (init → use → deinit → sleep → wake → init again) for battery saving.

**Questions:**
1. **Was this bug ever fixed** in newer SDK versions? (I'm using SDK 2.2.0)
2. **Is there a working workaround** to safely reinitialize `cyw43_arch` after `deinit()` for BLE use cases?

My current issue: first `cyw43_arch_init()` works, `cyw43_arch_deinit()` completes, but the second `cyw43_arch_init()` hangs.
Any guidance on how to properly handle BLE power cycling on Pico 2W would be greatly appreciated!
Thanks!

Memotech-Bill, 2026-02-15:
It is a while since I last tried this.

Also the CYW43 code has completely changed since I raised this issue, it now uses the pico/async_context. So any fix of the issue I raised would be irrelevant.

I also note that the current **cyw43_arch_init**  and **cyw43_arch_deinit** routines include code enclosed in CYW43_ENABLE_BLUETOOTH conditionals. Since I am not using Bluetooth I would not test that part of the code anyway.