raspberrypi/pico-sdk #982: stdio_flush() seems to do nothing and documentation is wrong

ecurtz, 2022-08-20:
As far as I can tell out_flush() isn't implemented in any of the stdio drivers, so stdio_flush() doesn't actually do anything.

Also the comment in the header file for stdio_flush() is copy/pasted from stdio_init_all()

wintersteiger, 2023-02-12:
+1. This is hugely annoying. I'm trying to log from both cores and multiple timers and no amount of flushing I do, the output is always some degree of garbled. For example:

```
auto_init_mutex(log_mtx);

void vllog(const char *fmt, va_list args) {
  mutex_enter_blocking(&log_mtx);
  printf("[%010llu] ", time_us_64());
  vprintf(fmt, args);
  printf("\n");
  fprintf(stdout, "\n");
  fflush(stdout);
  mutex_exit(&log_mtx);
}
```

None of `printf("\n");`, `fprintf(stdout, "\n");`, or `fflush(stdout);` makes sure that the output is in fact put out.

drankinatty, 2023-12-05:
Attempting to use it results in a `-Wimplicit-function-declaration` warning:

```none
/home/david/dev/arm/pico/src/fatfs-shell/tests/bench.c:210:9: warning: implicit declaration of function 'stdio_flush' [-Wimplicit-function-declaration]
  210 |         stdio_flush();
      |         ^~~~~~~~~~~
```

Which, along with the greyed out entry in [Pico-SDK](https://www.raspberrypi.com/documentation/pico-sdk/runtime.html#pico_stdio)  tends to confirm the lack of implementation.

peterharperuk, 2024-08-09:
I think this is fixed - can you retest?

closed 2024-08-09 by peterharperuk