raspberrypi/pico-sdk #983: Spurious #endif in arch_poll.h

Memotech-Bill, 2022-08-23:
There is a spurious #endif in file src/rp2_common/pico_cyw43_arch/include/pico/cyw43_arch/arch_poll.h at [line 44](https://github.com/raspberrypi/pico-sdk/blob/2e6142b15b8a75c1227dd3edbe839193b2bf9041/src/rp2_common/pico_cyw43_arch/include/pico/cyw43_arch/arch_poll.h#L44).

As a result, if this file is directly or indirectly included twice in a source file, `error: redefinition of 'cyw43_arch_lwip_protect'` is obtained on compilation.

For some reason the compiler is not complaining about the extra #endif at the end of the file.

fivdi, 2022-08-23:
Everything looks balanced to me (see it indented below), but it still looks like something is wrong with `#ifndef DOXYGEN_GENERATION` as `DOXYGEN_GENERATION` isn't `#define`d, so including the header twice means that `cyw43_arch_lwip_protect` will get redefined.
```c
/*
 * Copyright (c) 2022 Raspberry Pi (Trading) Ltd.
 *
 * SPDX-License-Identifier: BSD-3-Clause
 */

#ifndef _PICO_CYW43_ARCH_ARCH_POLL_H
    #define _PICO_CYW43_ARCH_ARCH_POLL_H

    #include "pico/cyw43_arch/arch_common.h"

    #include <assert.h>

    #ifdef __cplusplus
        extern "C" {
    #endif

    #define CYW43_THREAD_ENTER
    #define CYW43_THREAD_EXIT
    #ifndef NDEBUG

        void cyw43_thread_check(void);

        #define cyw43_arch_lwip_check() cyw43_thread_check()
        #define CYW43_THREAD_LOCK_CHECK cyw43_arch_lwip_check();
    #else
        #define cyw43_arch_lwip_check() ((void)0)
        #define CYW43_THREAD_LOCK_CHECK
    #endif

    #define CYW43_SDPCM_SEND_COMMON_WAIT cyw43_poll_required = true;
    #define CYW43_DO_IOCTL_WAIT cyw43_poll_required = true;

    #define cyw43_delay_ms sleep_ms
    #define cyw43_delay_us sleep_us

    void cyw43_schedule_internal_poll_dispatch(void (*func)(void));

    void cyw43_post_poll_hook(void);

    extern bool cyw43_poll_required;

    #define CYW43_POST_POLL_HOOK cyw43_post_poll_hook();
#endif

#ifndef DOXYGEN_GENERATION // multiple definitions in separate headers seems to confused doxygen
    #define cyw43_arch_lwip_begin() ((void)0)
    #define cyw43_arch_lwip_end() ((void)0)

    static inline int cyw43_arch_lwip_protect(int (*func)(void *param), void *param) {
        return func(param);
    }

    #ifdef __cplusplus
        }
    #endif

#endif
```

fivdi, 2022-08-23:
The way `#ifdef __cplusplus` is used in the header looks a bit dodgy. `#ifdef __cplusplus` is used once inside `
#ifndef _PICO_CYW43_ARCH_ARCH_POLL_H` and once inside `#ifndef DOXYGEN_GENERATION`

Memotech-Bill, 2022-08-23:
I agree with the previous comment. It looks as though the #endif is not spurious, but is in the wrong position. It should be as below.

```
/*
 * Copyright (c) 2022 Raspberry Pi (Trading) Ltd.
 *
 * SPDX-License-Identifier: BSD-3-Clause
 */

#ifndef _PICO_CYW43_ARCH_ARCH_POLL_H
    #define _PICO_CYW43_ARCH_ARCH_POLL_H

    #include "pico/cyw43_arch/arch_common.h"

    #include <assert.h>

    #ifdef __cplusplus
        extern "C" {
    #endif

    #define CYW43_THREAD_ENTER
    #define CYW43_THREAD_EXIT
    #ifndef NDEBUG

        void cyw43_thread_check(void);

        #define cyw43_arch_lwip_check() cyw43_thread_check()
        #define CYW43_THREAD_LOCK_CHECK cyw43_arch_lwip_check();
    #else
        #define cyw43_arch_lwip_check() ((void)0)
        #define CYW43_THREAD_LOCK_CHECK
    #endif

    #define CYW43_SDPCM_SEND_COMMON_WAIT cyw43_poll_required = true;
    #define CYW43_DO_IOCTL_WAIT cyw43_poll_required = true;

    #define cyw43_delay_ms sleep_ms
    #define cyw43_delay_us sleep_us

    void cyw43_schedule_internal_poll_dispatch(void (*func)(void));

    void cyw43_post_poll_hook(void);

    extern bool cyw43_poll_required;

    #define CYW43_POST_POLL_HOOK cyw43_post_poll_hook();

    #ifndef DOXYGEN_GENERATION // multiple definitions in separate headers seems to confused doxygen
        #define cyw43_arch_lwip_begin() ((void)0)
        #define cyw43_arch_lwip_end() ((void)0)

        static inline int cyw43_arch_lwip_protect(int (*func)(void *param), void *param) {
            return func(param);
        }
    #endif

    #ifdef __cplusplus
        }
    #endif

#endif

```

kilograham, 2023-01-18:
fixed by #1177

kilograham, 2023-01-24:
#1177 merged into develop

closed 2023-01-24 by kilograham