raspberrypi/pico-sdk #985: Possible Bug in I2C [SHTC3 Sensor Module]

MyVanitar, 2022-08-23:
Hello, I have successfully run the SHTC3 I2C sensor on STM32, but it does not work with PICO. I only replaced the STM32 HAL I2C commands with PICO SDK I2C commands. I should mention that I just have a problem with this sensor, other I2C staff which I have around work without any issue, but the reason for this weird problem is still unknown. The sensor itself is fine and works with other hardware. no hardware problem.

**Header File**

```c
#ifndef __SHTC3_H
#define __SHTC3_H
#include "pico/stdlib.h"

// The shtc3 provides a serial number individualized for each device
// Params:
//  - `hi2c` I2C bus
// Returns device id or 0 in case of error.
uint16_t shtc3_read_id();

// Put sensor into sleep mode
// Params:
//  - `hi2c` I2C bus
// Returns device id or 0 in case of error.
uint32_t shtc3_sleep();

// Wake up sensor.
// You must wait for 240us to let sensor enter into IDLE mode.
// Params:
//  - `hi2c` I2C bus
// Returns zero in case of error
uint32_t shtc3_wakeup();

// Performs full cycle: starts temperature/humidity measurements using "clock stretch" method.
// Params:
//  - `hi2c` I2C bus
//  - `temp` measured temperature, in C multiplied by 100 (e.g. 24.1C -> 2410)
//  - `hum` measured relative humidity, in percents
// Returns zero in case of error
uint32_t shtc3_perform_measurements();

// Start temperature/humidity measurements using "clock stretch" approach, in low power mode.
// After completed - values can be obtained by shtc3_read_measurements()
// Params:
//  - `hi2c` I2C bus
//  - `temp` measured temperature, in C multiplied by 100 (e.g. 24.1C -> 2410)
//  - `hum` measured relative humidity, in percents
// Returns zero in case of error
uint32_t shtc3_perform_measurements_low_power();

#endif

```

**Source**

```c
#include "shtc3.h"
#include "hardware/i2c.h"

#define SHTC3_ADDRESS_READ          (0x70 << 1) | 0x01
#define SHTC3_ADDRESS_WRITE         (0x70 << 1)

#define SHTC3_PRODUCT_CODE_MASK     0x083F
#define SHTC3_SENSOR_ID_MASK        0xF7C0

// NOTE: all commands are "byte swapped" (ntohs), meaning:
// 0x3517 -> 0x1735

#define SHTC3_CMD_WAKEUP                                    0x1735
#define SHTC3_CMD_SLEEP                                     0x98B0
#define SHTC3_CMD_SOFT_RESET                                0x5D80
#define SHTC3_CMD_READ_ID                                   0xC8EF

// Clock stretching based commands
#define SHTC3_CMD_CLK_STRETCH_READ_HUM_FIRST                0x245C
#define SHTC3_CMD_CLK_STRETCH_READ_HUM_FIRST_LOW_POWER      0xDE44

// Polling commands
#define SHTC3_CMD_POLL_HUM_FIRST                            0xE058
#define SHTC3_CMD_POLL_HUM_FIRST_LOW_POWER                  0x1A40

uint16_t shtc3_read_id()
{
  uint8_t data[2];
  int16_t command = SHTC3_CMD_READ_ID;

  i2c_write_blocking(i2c0, SHTC3_ADDRESS_WRITE, (uint8_t *)&command, 2, false);
  
  i2c_read_blocking(i2c0, SHTC3_ADDRESS_READ, (uint8_t *)data, 2, false);
  

  // SHTC3 16 bit ID encoded as:
  // xxxx 1xxx xx00 0111
  // where "x" are actual, sensor ID, while the rest
  // sensor product code (unchangeable)
  uint16_t id = data[0] << 8 | data[1];
  uint16_t code = id & SHTC3_PRODUCT_CODE_MASK;
  if (code == 0x807) {
    // Sensor preset, return actual ID
    return id & SHTC3_SENSOR_ID_MASK;
  }

  return 0;
}

uint32_t shtc3_sleep()
{
  uint16_t command = SHTC3_CMD_SLEEP;
  uint32_t res = i2c_write_timeout_us(i2c0, SHTC3_ADDRESS_WRITE, (uint8_t*)&command, 2, false , 5000);
  
}

uint32_t shtc3_wakeup()
{
  uint16_t command = SHTC3_CMD_WAKEUP;
  uint32_t res = i2c_write_timeout_us(i2c0, SHTC3_ADDRESS_WRITE, (uint8_t*)&command, 2, false, 5000);
  
}

static uint32_t checkCRC(uint16_t value, uint8_t expected)
{
	uint8_t data[2] = {value >> 8, value & 0xFF};
	uint8_t crc = 0xFF;
	uint8_t poly = 0x31;

	for (uint8_t indi = 0; indi < 2; indi++) {
		crc ^= data[indi];
		for (uint8_t indj = 0; indj < 8; indj++) {
			if (crc & 0x80) {
				crc = (uint8_t)((crc << 1) ^ poly);
			} else {
				crc <<= 1;
			}
		}
	}

	if (expected ^ crc)	{
    return 0;
	}
  return 1;
}

static uint32_t _read_values(uint8_t* data, int32_t* out_temp, int32_t* out_hum)
{
  // Check CRC
  uint32_t raw_hum = data[0] << 8 | data[1];
  uint32_t raw_temp = data[3] << 8 | data[4];

  if (!checkCRC(raw_hum, data[2])) {
    return 0;
  }
  if (!checkCRC(raw_temp, data[5])) {
    return 0;
  }

  // Convert values
  if (out_hum) {
    *out_hum = raw_hum * 100 / 65535;
  }
  if (out_temp) {
    *out_temp = raw_temp * 17500 / 65535 - 4500;
  }

  return 1;
}

static uint32_t _perform_measurements(uint16_t command, int32_t* out_temp, int32_t* out_hum)
{
  uint8_t result[6];

  uint32_t res = i2c_write_timeout_us(i2c0, SHTC3_ADDRESS_WRITE, (uint8_t*)&command, 2, false, 5000);
 
  res = i2c_read_timeout_us(i2c0, SHTC3_ADDRESS_READ, result, 6, false, 15000);
 
  return _read_values(result, out_temp, out_hum);
}

uint32_t shtc3_perform_measurements(int32_t* out_temp, int32_t* out_hum)
{
  return _perform_measurements(SHTC3_CMD_CLK_STRETCH_READ_HUM_FIRST, out_temp, out_hum);
}

uint32_t shtc3_perform_measurements_low_power(int32_t* out_temp, int32_t* out_hum)
{
  return _perform_measurements(SHTC3_CMD_CLK_STRETCH_READ_HUM_FIRST_LOW_POWER, out_temp, out_hum);
}

```

**To run**

```c
shtc3_perform_measurements(&temperature, &humidity);
```

lurch, 2022-08-24:
I suspect that the Raspberry Pi engineers are all too busy to help with debugging random I2C devices? So you may be better off asking for help at https://forums.raspberrypi.com/viewforum.php?f=143

If you _do_ want help from a Raspberry Pi engineer, you'll probably want to provide logic-analyser traces for both the working (STM32) and non-working (RP2040) I2C transactions, so that they can see what any differences might be.

MyVanitar, 2022-08-24:
Alright, I captured the signal using an oscilloscope. in both cases, I just call the function `shtc3_read_id()`

**STM32**
[image]

**PICO**
[image]

czietz, 2022-08-24:
Why are you shifting the addresses by one?
```c
#define SHTC3_ADDRESS_READ          (0x70 << 1) | 0x01
#define SHTC3_ADDRESS_WRITE         (0x70 << 1)
```
Have you tried _not_ doing that? Pico SDK takes care of setting the R/W bit for you. Therefore, you have to pass the _7 bit_ address to the SDK functions:
> i2c_write_blocking
> [...]
> addr: 7-bit address of device to write to

MyVanitar, 2022-08-24:
> Why are you shifting the addresses by one?

I tried your suggestion, and wow, thank you very much, how did you know that? I read everywhere
I read everywhere, but I was not aware of your point, below worked, so as you said, no need to differentiate the address for read and write

```c
#define SHTC3_ADDRESS_READ          (0x70) 
#define SHTC3_ADDRESS_WRITE         (0x70)
```

lurch, 2022-08-24:
Ah great, so this can be closed now?

closed 2022-08-24 by MyVanitar

czietz, 2022-08-24:
> I tried your suggestion, and wow, thank you very much, how did you know
> that?

Perhaps a little bit of background info helps: I2C (usually) has 7-bit
addresses. When starting a transaction, the address is immediately
followed by a bit that indicates a read or write command. Apparently,
with your previous hardware, you had to include this bit in the address.
This is why you had the left shift (and "| 0x01" in case of the read).

However, with the Pico SDK, and as described in its documentation, you
only pass the 7-bit address of the I2C device. The Pico SDK already
knows from the function call whether it is a read or write command.