raspberrypi/pico-sdk #988: pio_sm_set_pindirs_with_mask misbehaves if OUT_STICKY is set

jannic, 2022-08-25:
We fixed an issue in the rust rp2040 support which is also present in the C SDK: https://github.com/rp-rs/rp-hal/pull/432

If the OUT_STICKY bit is set, pio_sm_set_pindirs_with_mask and similar functions will change the out or outdir values of unrelated pins.
The values written using the injected SET instruction will be sticky, ie. they will be applied again each PIO cycle.
This still happens after resetting `pinctrl` with `pio->sm[sm].pinctrl = pinctrl_saved`. At this point, depending on the previous value of `pinctrl`, the values will be written to some other unintended pin.

kilograham, 2022-08-25:
assign to @Wren6991 to think about

kilograham, 2023-05-26:
was fixed in 1.5.0

PR #989 (2022-10-16): Make sure OUT_STICKY is not set when setting pins / pindirs
To avoid the issue described in #988, temporarily clear the OUT_STICKY bit in the affected functions, and restore the value later.
(Note that I didn't test this fix with the C SDK. It's a port of https://github.com/rp-rs/rp-hal/pull/432 which was tested on real hardware, but I might have introduced bugs when porting it to C - my C has become quite rusty, pun intended... Please review carefully)

Fixes #988
closed 2023-05-26 by kilograham