raspberrypi/pico-sdk #996: ERROR : Pico second stage bootloader

Elyonni, 2022-08-30:
Hello, 
I am a novice and I am building a project using FreeRtos. So I've oriented the Raspberry pi pico as the micro controller on which my executable will run.
I'm trying to build my own cmake file because even if the SDK is very well done it does too many things so I don't need (and I don't understand everything). So I manage to generate an elf file and then I use the elf2uf2 tool to convert the uf2 file. And this is where the memory problems start :) 
I understood that we have to link the boot_stage2.ld file and then memmap_default.ld file but it doesn't work ( ERROR: Pico second stage bootloader must be 256 bytes in size warning: cannot find entry symbol _entry_point; defaulting to 0000000010000000) . I don't understand what I am missing. 
Could you please help me?

kilograham, 2022-08-30:
You should ask questions like this on the forums https://forums.raspberrypi.com/viewforum.php?f=143

closed 2022-08-30 by kilograham

jleaders, 2023-02-12:
The error "ERROR: Pico second stage bootloader must be 256 bytes in size" is typically encountered when you are trying to link a Pico RP2040 firmware image using the linker (ld) and the size of the image exceeds 256 bytes.

The RP2040 microcontroller has a second stage bootloader that is used to load and execute firmware images. This bootloader is located in a specific region of the flash memory and has a fixed size of 256 bytes. If the size of the linked firmware image exceeds 256 bytes, the bootloader will be overwritten and the device will be unable to boot.

To resolve this issue, you need to ensure that the size of the firmware image does not exceed 256 bytes. This can be achieved by reducing the size of the image itself, or by excluding certain sections of the image that are not needed.

Here are a few steps that you can follow to reduce the size of your firmware image:

Use linker scripts: Linker scripts allow you to control the layout of the firmware image in memory, and can be used to exclude sections of the image that are not needed. You can use the --script option to specify the linker script to use when linking the image.

Optimize your code: You can reduce the size of the image by optimizing your code. This can be achieved by removing unnecessary functions, variables, and data structures, as well as using more efficient algorithms and data structures.

Reduce the size of static data: Static data, such as arrays and global variables, can take up a significant amount of space in the firmware image. You can reduce the size of static data by using smaller data types, such as uint8_t instead of int, and by initializing arrays with only the values that you need.

Compile with optimizations: Compiling your code with optimizations can also help to reduce the size of the firmware image. You can enable optimizations by using the -O option when compiling your code.

By following these steps, you can reduce the size of your firmware image and resolve the "ERROR: Pico second stage bootloader must be 256 bytes in size" error.