raspberrypi/pico-sdk #1002: USB not recognised when dropping uf2 file on Pico.

FraserElectronics, 2022-09-01:
I am trying to develop an app on the Pico W using the Pico W SDK.
I created a simple program, built it and copied it to the Pico W drive.
The Pico reboots then after about 10-15 seconds I get a device not recognised.
Part of my application need to send and receive data over the USB but I can't do that unless it enumerates as a serial port on Windows 10.
There seems to be plenty of hits with people having similar problems but I can't find any solution that works for me.
Does anyone have any advice ?

Andy

lurch, 2022-09-01:
I think this is a question for the forums https://forums.raspberrypi.com/viewforum.php?f=143

FraserElectronics, 2022-09-01:
Andrew,

Thanks - I have posted under the Pico->SDK on the Raspberry Pi Forum.

Regards

Andy

closed 2022-09-01 by FraserElectronics