raspberrypi/pico-sdk #999: Unable to build when using `pico_cyw43_arch_lwip_sys_freertos`

stackjohn, 2022-09-01:
Lets say I have the following cmake config:

```
target_link_libraries(
  main
  pico_stdlib
  cyw43_driver_base
  pico_cyw43_arch_lwip_threadsafe_background
  pico_lwip_http
  hardware_adc
  freertos
)
```

This all builds fine. I can then start the httpd server in it's own freertos task:

```
void pico_w_http_task()
{
    httpd_init();
    ssi_init();
    printf("Http server initialized.\n");
    for (;;)
    {
        vTaskDelay(pdMS_TO_TICKS(10));
    }
}
```

```
    xTaskCreate(pico_w_http_task, "HTTP_Task", 4000, NULL, 1, NULL);
```

Not really sure if this is the correct way to do things however 😬 
It works ok but sometimes it seems to lock up the Pico when making a number of concurrent http requests. 

The release notes make a mention of `pico_cyw43_arch_lwip_sys_freertos` to replace `pico_cyw43_arch_lwip_threadsafe_background`.  Trying this in my cmake config:

```
target_link_libraries(
  main
  pico_stdlib
  cyw43_driver_base
  pico_cyw43_arch_lwip_sys_freertos
  pico_lwip_http
  hardware_adc
  freertos
)
```

I get the following errors:

```
[ 12%] Building C object CMakeFiles/reboot_post.dir/pico-sdk/lib/lwip/contrib/ports/freertos/sys_arch.c.obj
/pico-sdk/lib/lwip/contrib/ports/freertos/sys_arch.c: In function 'sys_arch_msleep':
/pico-sdk/lib/lwip/contrib/ports/freertos/sys_arch.c:192:39: error: 'portTICK_RATE_MS' undeclared (first use in this function); did you mean 'portTICK_PERIOD_MS'?
  192 |   TickType_t delay_ticks = delay_ms / portTICK_RATE_MS;
      |                                       ^~~~~~~~~~~~~~~~
      |                                       portTICK_PERIOD_MS
/pico-sdk/lib/lwip/contrib/ports/freertos/sys_arch.c:192:39: note: each undeclared identifier is reported only once for each function it appears in
/pico-sdk/lib/lwip/contrib/ports/freertos/sys_arch.c: In function 'sys_arch_sem_wait':
/pico-sdk/lib/lwip/contrib/ports/freertos/sys_arch.c:294:45: error: 'portTICK_RATE_MS' undeclared (first use in this function); did you mean 'portTICK_PERIOD_MS'?
  294 |     TickType_t timeout_ticks = timeout_ms / portTICK_RATE_MS;
      |                                             ^~~~~~~~~~~~~~~~
      |                                             portTICK_PERIOD_MS
/pico-sdk/lib/lwip/contrib/ports/freertos/sys_arch.c: In function 'sys_arch_mbox_fetch':
/pico-sdk/lib/lwip/contrib/ports/freertos/sys_arch.c:401:45: error: 'portTICK_RATE_MS' undeclared (first use in this function); did you mean 'portTICK_PERIOD_MS'?
  401 |     TickType_t timeout_ticks = timeout_ms / portTICK_RATE_MS;
      |                                             ^~~~~~~~~~~~~~~~
      |                                             portTICK_PERIOD_MS
make[2]: *** [CMakeFiles/reboot_post.dir/pico-sdk/lib/lwip/contrib/ports/freertos/sys_arch.c.obj] Error 1
make[1]: *** [CMakeFiles/reboot_post.dir/all] Error 2
make: *** [all] Error 2
```

How can I resolve these errors? Are there any docs around how to use pico_lwip_http when using `  pico_cyw43_arch_lwip_sys_freertos`? Such as should it be started in it's own thread or started independently during main?

peterharperuk, 2022-09-01:
Have you looked at the freertos sdk examples? It looks like you have a configuration issue. Do the examples build? https://github.com/raspberrypi/pico-examples/tree/master/pico_w/freertos

peterharperuk, 2022-09-01:
A quick play and this worked for me https://github.com/peterharperuk/pico-examples/commit/9e63d6daf8e375b12a157746482e5f0afcc5f6d0

stackjohn, 2022-09-01:
Thanks @peterharperuk 

This single line seems to allow it to build, set in `FreeRTOSConfig.h`:

```
#define configENABLE_BACKWARD_COMPATIBILITY     1
```

But then the pico just seems to fail to boot. 

I also tried with just this, also in `FreeRTOSConfig.h`: 

```
#define portTICK_RATE_MS portTICK_PERIOD_MS
```

It builds, but again the pico just seems to lockup, pretty instantly, I get no output via usb even when just sitting in a busy_wait loop for a while after running `stdio_init_all();`

I'm using the SMP branch of the `FreeRtos_kernel` repo, not sure if that makes any difference. 

This is how `FreeRTOSConfig.h` looks:

```
/*
 * FreeRTOS V202104.00
 * Copyright (C) 2020 Amazon.com, Inc. or its affiliates.  All Rights Reserved.
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy of
 * this software and associated documentation files (the "Software"), to deal in
 * the Software without restriction, including without limitation the rights to
 * use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of
 * the Software, and to permit persons to whom the Software is furnished to do so,
 * subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all
 * copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
 * FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
 * COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
 * IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
 * CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
 * http://www.FreeRTOS.org
 * http://aws.amazon.com/freertos
 *
 * 1 tab == 4 spaces!
 */

#ifndef FREERTOS_CONFIG_H
#define FREERTOS_CONFIG_H

/*-----------------------------------------------------------
 * Application specific definitions.
 *
 * These definitions should be adjusted for your particular hardware and
 * application requirements.
 *
 * THESE PARAMETERS ARE DESCRIBED WITHIN THE 'CONFIGURATION' SECTION OF THE
 * FreeRTOS API DOCUMENTATION AVAILABLE ON THE FreeRTOS.org WEB SITE.
 *
 * See http://www.freertos.org/a00110.html
 *----------------------------------------------------------*/

/* Scheduler Related */
#define configUSE_PREEMPTION 1
#define configUSE_TICKLESS_IDLE 0
#define configUSE_IDLE_HOOK 0
#define configUSE_TICK_HOOK 0
#define configTICK_RATE_HZ ((TickType_t)1000)
#define configMAX_PRIORITIES 32
#define configMINIMAL_STACK_SIZE (configSTACK_DEPTH_TYPE)256
#define configUSE_16_BIT_TICKS 0

#define configIDLE_SHOULD_YIELD 1

/* Synchronization Related */
#define configUSE_MUTEXES 1
#define configUSE_RECURSIVE_MUTEXES 1
#define configUSE_APPLICATION_TASK_TAG 0
#define configUSE_COUNTING_SEMAPHORES 1
#define configQUEUE_REGISTRY_SIZE 8
#define configUSE_QUEUE_SETS 1
#define configUSE_TIME_SLICING 1
#define configUSE_NEWLIB_REENTRANT 0
#define configENABLE_BACKWARD_COMPATIBILITY 0
#define configNUM_THREAD_LOCAL_STORAGE_POINTERS 5

#define portTICK_RATE_MS portTICK_PERIOD_MS
/* System */
#define configSTACK_DEPTH_TYPE uint32_t
#define configMESSAGE_BUFFER_LENGTH_TYPE size_t

/* Memory allocation related definitions. */
#define configSUPPORT_STATIC_ALLOCATION 0
#define configSUPPORT_DYNAMIC_ALLOCATION 1
#define configTOTAL_HEAP_SIZE (128 * 1024)
#define configAPPLICATION_ALLOCATED_HEAP 0

/* Hook function related definitions. */
#define configCHECK_FOR_STACK_OVERFLOW 0
#define configUSE_MALLOC_FAILED_HOOK 0
#define configUSE_DAEMON_TASK_STARTUP_HOOK 0

/* Run time and task stats gathering related definitions. */
#define configGENERATE_RUN_TIME_STATS 0
#define configUSE_TRACE_FACILITY 1
#define configUSE_STATS_FORMATTING_FUNCTIONS 0

/* Co-routine related definitions. */
#define configUSE_CO_ROUTINES 0
#define configMAX_CO_ROUTINE_PRIORITIES 1

/* Software timer related definitions. */
#define configUSE_TIMERS 1
#define configTIMER_TASK_PRIORITY (configMAX_PRIORITIES - 1)
#define configTIMER_QUEUE_LENGTH 10
#define configTIMER_TASK_STACK_DEPTH 1024

/* Interrupt nesting behaviour configuration. */
/*
#define configKERNEL_INTERRUPT_PRIORITY         [dependent of processor]
#define configMAX_SYSCALL_INTERRUPT_PRIORITY    [dependent on processor and application]
#define configMAX_API_CALL_INTERRUPT_PRIORITY   [dependent on processor and application]
*/

/* SMP port only */
#define configNUM_CORES 2
#define configTICK_CORE 1
#define configRUN_MULTIPLE_PRIORITIES 1
#define configUSE_CORE_AFFINITY 1

/* RP2040 specific */
#define configSUPPORT_PICO_SYNC_INTEROP 1
#define configSUPPORT_PICO_TIME_INTEROP 1

#include <assert.h>
/* Define to trap errors during development. */
#define configASSERT(x) assert(x)

/* Set the following definitions to 1 to include the API function, or zero
to exclude the API function. */
#define INCLUDE_vTaskPrioritySet 1
#define INCLUDE_uxTaskPriorityGet 1
#define INCLUDE_vTaskDelete 1
#define INCLUDE_vTaskSuspend 1
#define INCLUDE_vTaskDelayUntil 1
#define INCLUDE_vTaskDelay 1
#define INCLUDE_xTaskGetSchedulerState 1
#define INCLUDE_xTaskGetCurrentTaskHandle 1
#define INCLUDE_uxTaskGetStackHighWaterMark 1
#define INCLUDE_xTaskGetIdleTaskHandle 1
#define INCLUDE_eTaskGetState 1
#define INCLUDE_xTimerPendFunctionCall 1
#define INCLUDE_xTaskAbortDelay 1
#define INCLUDE_xTaskGetHandle 1
#define INCLUDE_xTaskResumeFromISR 1
#define INCLUDE_xQueueGetMutexHolder 1

/* A header file that defines trace macro can be included here. */

#endif /* FREERTOS_CONFIG_H */
```f

peterharperuk, 2022-09-01:
>> I'm using the SMP branch of the FreeRtos_kernel repo, not sure if that makes any difference.
Can you try using main?

Does this work? https://github.com/peterharperuk/pico-examples/commit/9e63d6daf8e375b12a157746482e5f0afcc5f6d0

stackjohn, 2022-09-01:
> > > I'm using the SMP branch of the FreeRtos_kernel repo, not sure if that makes any difference.
> > > Can you try using main?
> 
> Does this work? [peterharperuk/pico-examples@9e63d6d](https://github.com/peterharperuk/pico-examples/commit/9e63d6daf8e375b12a157746482e5f0afcc5f6d0)

Can you explain how to build it? I tried:

```
mkdir build
cd build
cmake ..
make
```

I get plenty of *.uf2 files but nothing for the pico_w.

JamesH65, 2022-09-01:
cmake .. -DPICO_BOARD=pico_w

?

peterharperuk, 2022-09-01:
cmake .. -DPICO_BOARD=pico_w -DWIFI_SSID='<wifi ssid name>' -DWIFI_PASSWORD=<wifi password> -DFREERTOS_KERNEL_PATH=<path to FreeRTOS-Kernel>

stackjohn, 2022-09-01:
Thanks @JamesH65 @peterharperuk 

I managed to get it to build. That works fine for me.... must be some wrong config somewhere in my code. 

Will try and figure it out. Thanks for the help!

stackjohn, 2022-09-01:
Going to close this..........  I think I had issues with my cmakelists file, it's all working fine now. Thanks again for the help!

closed 2022-09-01 by stackjohn