raspberrypi/pico-sdk #1003: How to handle wifi client disconnection ?

Entretoize, 2022-09-02:
I'm using ap mode and connect to my pico from my pc, the problem is that when the pc disconnect from the pico, the pico keeps waiting for the tcp client to send data.
I added a timeout to the client socket but there's no difference between a client sending nothing and a client that left the wifi, then I can't break the loop.
Is there a way to check if the client is still connected ?

My code:

```
rp2.country('FR')
ap = network.WLAN(network.AP_IF)
ap.config(essid=ssid, password=password)
ap.active(True)

while ap.active() == False:
  pass

addr = socket.getaddrinfo('0.0.0.0', 80)[0][-1]
s = socket.socket()
s.bind(addr)
s.listen(1)

while True:
    try:
        print("waiting client...")
        cl, addr = s.accept()
        cl.settimeout(1.0)
        print("Client connected from", addr)
        
        while True:           
            try:
                r = cl.recv(1024)

            except OSError as e:
                print(e)
                continue
                    
           #some code to handle tcp messages

        cl.close()
        
        print("close")

    except OSError as e:
        cl.close()
        print('connection closed on error',e)
```

peterharperuk, 2022-09-02:
This isn't really a pico-sdk issue? You might get a quicker answer from the forum [micropython](https://forums.raspberrypi.com/viewforum.php?f=146).

kilograham, 2022-09-02:
not an SDK issue

closed 2022-09-02 by kilograham