raspberrypi/pico-sdk #1004: Is the function `uart_get_instance` part of the official API?

steverpalmer, 2022-09-04:
The "hardware/uart.h" include file contains the function `uart_get_instance` to map from a `uint` to a `uart_inst_t *`.  However, it is not listed on the API [webpage](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__uart.html) as part of the API.

For Your Information, the TinyUSB github code does make use of `uart_get_instance` it's [rp2040 bsp](https://github.com/hathach/tinyusb/blob/master/hw/bsp/rp2040/family.c), so in some sense it has already "escaped".

I suspect that `uart_get_instance` is a drop-off from the Web page, but I get twitchy about using features that are not part of the official API.  So, is the function `uart_get_instance` meant to be part of the official API?  Should it be listed on the Webpage?

kilograham, 2022-09-06:
yes it is part of the public API

steverpalmer, 2022-09-07:
@kilograham - Thanks for the prompt reply.  I'll go ahead and use it.

Housekeeping: I don't know whether `uart_get_instance` should be listed on the Webpage, and this issue has been tagged which might indicate that further work is indicated, so I won't close this issue myself.  For whatever it's worth, I am content for it to be closed.

lurch, 2022-09-07:
Yes, I believe Graham added the `documentation` label as a reminder to himself that he needs to add appropriate Doxygen comments to https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_uart/include/hardware/uart.h#L105 , which is what causes it to "appear on the webpage" that you're referring to.

peterharperuk, 2024-08-09:
This appears to be fixed. I can see it in the docs.

closed 2024-08-09 by peterharperuk