raspberrypi/pico-sdk #1006: stdio_usb drops stdout from printf calls and somehow emits garbage memory to the terminal window

JamesDunne, 2022-09-04:
See https://github.com/JamesDunne/pico-stdio-usb-bug for a minimal(-ish) reproducible example with no external dependencies required.

Steps to reproduce:
1. Use Pico board with USB connection to host PC
2. Build example repo
3. Flash firmware
4. Connect PuTTY (or KiTTY) using serial mode to COM5 (or wherever your Pico is)
5. Observe output until seeing dropped data and/or corrupted terminal output

Example output:
[image]

The output text above indicates some sort of memory overflow bug in KiTTY but their bug should not be triggered by such trivial ASCII-only `printf()` calls from the Pico. The presence of text like `RegDeleteValue` clearly indicates a Windows-side memory overflow since none of that text should exist in the Pico binaries.

JamesDunne, 2022-09-04:
Commenting out the only use of format-specifiers in printf() calls triggers the issue far less often but still triggers it.

[image]

KubaO, 2022-09-29:
> their bug should not be triggered by such trivial ASCII-only printf() calls from the Pico

Why not? They have a bug, and a bad one. This probably could be exploited. This should be reported to KiTTY as an urgent issue, since it has safety implications.

kilograham, 2023-05-26:
no evidence of this being anything other than a Pu/KiTTY bug

closed 2023-05-26 by kilograham