raspberrypi/pico-sdk #1007: Incorrect argument to gpio_add_raw_irq_handler_with_order_priority() in pico_cyw43_arch

eskimo-software, 2022-09-06:
The following files in pico_cyw43_arch:
  
  [cyw43_arch_poll.c](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/pico_cyw43_arch/cyw43_arch_poll.c)
  [cyw43_arch_[yw43_arch_threadsafe_background.c](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/pico_cyw43_arch/cyw43_arch_threadsafe_background.c).c](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/pico_cyw43_arch/cyw43_arch_[yw43_arch_threadsafe_background.c](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/pico_cyw43_arch/cyw43_arch_threadsafe_background.c).c)
  [cyw43_arch_freertos.c](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/pico_cyw43_arch/cyw43_arch_freertos.c)

each install a GPIO interrupt handler via the following call:

`    gpio_add_raw_irq_handler_with_order_priority(IO_IRQ_BANK0, gpio_irq_handler, CYW43_GPIO_IRQ_HANDLER_PRIORITY);`

The first argument IO_IRQ_BANK0 identifies the IRQ number for the GPIO interrupt - but the function actually expects the specific GPIO number for which to install the handler.

i.e. The argument IO_IRQ_BANK0 should be replaced with CYW43_PIN_WL_HOST_WAKE

lurch, 2022-09-06:
ping @peterharperuk

envirocoder, 2022-10-28:
Fixed by #1069

kilograham, 2023-01-18:
should be fixed by #1177

kilograham, 2023-01-24:
merged into develop

closed 2023-01-24 by kilograham