raspberrypi/pico-sdk #1008: TinyUSB print to console

0i9j, 2022-09-07:
Is there a function that does the same like `printf()` in the TinyUSB library? I would use `printf()` but i need my code to run as fast as possible so i am looking for an alternative to write 16Bit Hex Values to e.g. COM19, where my Raspberry Pico is connected, on my PC. I want to see the values in Putty to confirm they´re correct. 
I tried `write()` but that prints the value in 8Bit ASCII with Endians mixed up. So 0x2F3D becomes 0x3D 0x2F.
I got the TinyUSB library suggested, but couldn´t really figure out how to use it properly. Is there a documentation other than the one on git and on their website?

lurch, 2022-09-07:
This sounds like a question for the forums https://forums.raspberrypi.com/viewforum.php?f=143

> I tried write() but that prints the value in 8Bit ASCII with Endians mixed up. So 0x2F3D becomes 0x3D 0x2F.

Did you try either 
```
write((your_var >> 8) & 0xff);
write(your_var & 0xff);
```
or
```
write(your_var & 0xff);
write((your_var >> 8) & 0xff);
```
?
Alternatively it shouldn't be hard to write your own function to format 16-bit integers as ASCII hex values. (I did something similar before to format 32-bit integers as ASCII hex values)

0i9j, 2022-09-07:
Thanks @lurch. There is not enough time to use `write()` twice. Whats interesting is, that as long i don´t open the COM19 in Putty the runtime of my function is about 2us, but when putty is connected and shows the values, the runtime shoots up to 4us. Probably makes sense because of the USB protocoll or sth like that? 
I could convert it to ASCII, that would fix the Endian problem, but e.g. 0x2f3D would still print to `0x2f 0x3D` on the output, if I would use `write()`. Gonna ask in the forum for sure.