raspberrypi/pico-sdk #1012: pico_add_extra_outputs doesn't follow RUNTIME_OUTPUT_DIRECTORY

sagarreddypatil, 2022-09-08:
I'm using a single CMake file for a multi-board project along with a libraries folder. This is what my main CMake File looks like:
```cmake
cmake_minimum_required(VERSION 3.13)

# ---------- Setup  ----------
# to make sure stuff in the cmake folder can be located
set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_SOURCE_DIR}/cmake/")

# Importing the pico-sdk importer
set(PICO_SDK_FETCH_FROM_GIT on)
include(pico_sdk_import)

# General utility macros
include(util)

# Project
project(CMS_Avionics_Code)
pico_sdk_init()

set(stdlibs pico_stdlib hardware_gpio hardware_spi)

# To make sure build folder doesn't get cluttered
set(CMAKE_ARCHIVE_OUTPUT_DIRECTORY ${CMAKE_BINARY_DIR}/lib)
set(CMAKE_LIBRARY_OUTPUT_DIRECTORY ${CMAKE_BINARY_DIR}/lib)

# ---------- Libraries  ----------
subdirlist(libs ${CMAKE_SOURCE_DIR}/lib)

foreach(libname ${libs})
    file(GLOB_RECURSE srcs ${CMAKE_SOURCE_DIR}/lib/${libname}/*.c)
    file(GLOB_RECURSE hdrs ${CMAKE_SOURCE_DIR}/lib/${libname}/*.h)

    add_library(${libname} ${srcs} ${hdrs})
    target_link_libraries(${libname} ${stdlibs})

    target_include_directories(${libname} PUBLIC ${CMAKE_SOURCE_DIR}/lib/${libname})
endforeach()

# # ---------- Boards  ----------
set(boards
    emu
)

foreach(board ${boards})
    file(GLOB_RECURSE srcs ${CMAKE_SOURCE_DIR}/${board}/*.c)
    file(GLOB_RECURSE hdrs ${CMAKE_SOURCE_DIR}/${board}/*.h)

    add_executable(${board} ${srcs} ${hdrs})
    target_link_libraries(${board} ${stdlibs} ${libs})

    pico_enable_stdio_usb(${board} 1)
    pico_enable_stdio_uart(${board} 0)

    pico_add_extra_outputs(${board})

    set_target_properties(${board}
        PROPERTIES

        ARCHIVE_OUTPUT_DIRECTORY ${CMAKE_BINARY_DIR}/${board}
        LIBRARY_OUTPUT_DIRECTORY ${CMAKE_BINARY_DIR}/${board}
        RUNTIME_OUTPUT_DIRECTORY ${CMAKE_BINARY_DIR}/${board}
    )
endforeach()
```

Then I run the following commands to build
```bash
cd build
cmake ..
make
```

However, only the .elf file for emu is present in `<project-root>/build/emu`, and the rest of the extra output files(.hex, .uf2, etc.) are present in `<project-root>/build/`

I am new to CMake, so I am not sure if I am doing something wrong. I'm opening this issue because it seems to be specific to the pico_add_extra_outputs function.

sagarreddypatil, 2022-09-10:
@kilograham Saw you on the Git Blame so I'm pinging you here.

sagarreddypatil, 2022-09-20:
@lurch ? Is this the wrong place to ask this question

lurch, 2022-09-20:
Nope, this *is* the right place to ask the question, but we're all very busy and so can't always reply immediately to every single GitHub issue.
For more "general" questions you can ask at https://forums.raspberrypi.com/viewforum.php?f=143 where you're more likely to get a quicker reply.

kilograham, 2022-10-21:
merged into develop

commit a88baafc (2022-10-21): Implemented ARCHIVE_OUTPUT_DIRECTORY for uf2 output in CMake. Fixes #1012 (#1036)

* Added ARCHIVE_OUTPUT_DIRECTORY to uf2 output in CMake

This Commit enables the use of the ARCHIVE_OUTPUT_DIRECTORY target
properties in CMake for the generation of uf2 files.

The changeset in lines 47..51 is necessary due to CMake not
automatically creating the ARCHIVE_OUTPUT_DIRECTORY if no archive target
is present.

* rework cmake changes to make it friendler for non absolute paths
closed 2022-10-21 by kilograham