raspberrypi/pico-sdk #1017: C++ compiler warning: compound assignment with 'volatile'-qualified left operand is deprecated

jonathangjertsen, 2022-09-10:
Including `hardware/pio.h` from a C++ file generates warnings like the following. Compiler version is compiler version: `arm-none-eabi-gcc (GNU Toolchain for the Arm Architecture 11.2-2022.02 (arm-11.16)) 11.2.1 20220111`.

```
In file included from <snip>/ws2812.cpp:1:
<snip>/pio.h: In function 'void pio_sm_restart(PIO, uint)':
<snip>/pio.h:583:15: error: compound assignment with 'volatile'-qualified left operand is deprecated [-Werror=volatile]
  583 |     pio->ctrl |= 1u << (PIO_CTRL_SM_RESTART_LSB + sm);
      |     ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
<snip>/pio.h: In function 'void pio_restart_sm_mask(PIO, uint32_t)':
<snip>/pio.h:598:15: error: compound assignment with 'volatile'-qualified left operand is deprecated [-Werror=volatile]
  598 |     pio->ctrl |= (mask << PIO_CTRL_SM_RESTART_LSB) & PIO_CTRL_SM_RESTART_BITS;
<snip>/pio.h: In function 'void pio_sm_clkdiv_restart(PIO, uint)':

... and so on.
```

This is because operations such as `|=` and `&=` on volatile variables is deprecated in C++20. Here is a summary of the change: https://blog.feabhas.com/2021/05/modern-embedded-c-deprecation-of-volatile/

What the C++ committee wants you to do is to rewrite each such statement as follows:

```diff
- pio->ctrl |= 1u << (PIO_CTRL_SM_RESTART_LSB + sm);
+ pio->ctrl = pio->ctrl | (1u << (PIO_CTRL_SM_RESTART_LSB + sm));
```

Sample of responses from other projects:

- [Zephyr](https://github.com/zephyrproject-rtos/zephyr/pull/30868) silenced the warning by adding `-Wno-volatile`, pointing to the fact that they do not control the CMSIS header files.
- [CMSIS](https://github.com/ARM-software/CMSIS_5/issues/1544) will not fix the issue in their files as they do not  officially support C++. The maintainer's suggestion that it should work with `extern "C"` is not correct since that's just a linkage directive. 
- [esp-idf](https://github.com/espressif/esp-idf/issues/9170) updated some macros which generated this warning.
- [STMCubeWB](https://github.com/STMicroelectronics/STM32CubeWB/issues/22) has not done anything

jonathangjertsen, 2022-09-10:
I tried compiling a C++ version of `kitchen_sink.c` and found that only pio.h and interp.h are affected, as those are the only files which perform that kind of operation. I will make a PR to make them C++20 compliant

geurtv, 2022-09-10:
This has been de-deprecated in C++23, so in my opinion the code should be left unchanged. Instead add -Wno-volatile to suppress this warning (until GCC13 is available). See [P2327R1: De-deprecating volatile compound operations](https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2021/p2327r1.pdf ).

jonathangjertsen, 2022-09-10:
> This has been de-deprecated in C++23, so in my opinion the code should be left unchanged. Instead add -Wno-volatile to suppress this warning (until GCC13 is available). See [P2327R1: De-deprecating volatile compound operations](https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2021/p2327r1.pdf).

Ah, that's great, I thought it was a strange decision. @maintainers please feel free to decline #1018 and add the flag in the appropriate place

kilograham, 2022-10-21:
merged into develop

PR #1018 (2022-10-16): Avoid compound assignment with volatile left operand as described in #1017
Fixes #1017

Checked by compiling a C++ file that includes all the same files included by `kitchen_sink.c`.
closed 2022-10-21 by kilograham