raspberrypi/pico-sdk #1025: Cannot initialize multiple stdio UART driver instances

alastairpatrick, 2022-09-13:
The API does not prevent an attempt to initialize multiple UART stdio driver instances. I can't find any documentation saying this isn't allowed. It would also be possible to implement. For example:
```c
int main() {
  stdio_uart_init_full(uart0, 115200, 0, 1);
  stdio_uart_init_full(uart1, 115200, 6, 7);

  printf("Hello world!\r\n");

  for (;;);
}
````
The expectation here might be that the greeting is output to both UARTs.

This could be fixed by changing the documentation and maybe adding a runtime assertion or by actually making it work.

If it's going to work, issue #811 is impacted because the signature of stdio_uart_deinit_full() should change to identify which UART driver instance is uninitialized:
```c
void stdio_uart_deinit_full(uart_inst_t* instance, int tx_pin, int rx_pin);
```

kilograham, 2023-05-26:
Having `uart0_handle` and `uart1_handle` is quite a nice idea

ekholm, 2024-09-18:
May I suggest to use the more standard handles, stdout and stderr, stdin. 
And in startup code application can initialize which uart(or usb) is what pipe

kilograham, 2024-09-18:
> Somewhat more involved, it would be useful if something like this worked:
>  ``` stdio_uart_init_full(uart0, 115200, 0, 1);
>  stdio_uart_init_full(uart1, 115200, 6, 7);
>
>  fprintf(uart0_handle, "Hello from uart0!\r\n");
>
>  fprintf(uart1_handle, "Hello from uart1!\r\n");
>```

See #1715 which would add support for more handles

kilograham, 2024-09-18:
> May I suggest to use the more standard handles, stdout and stderr, stdin. And in startup code application can initialize which uart(or usb) is what pipe

I think @alastairpatrick 's point was that the `uartn_` handles could be used in addition to `stdout` etc.  for finer grained control