raspberrypi/pico-sdk #1027: Wrong CMSIS mpu definitions?

WestfW, 2022-09-17:
as part of the CMSIS add, there exists a file:
.../src/rp2_common/cmsis/stub/CMSIS/Core/Include/mpu_armv7.h
containing the memory protection unit definitions for "ARM v7m" cpus.

But the rp2040 is a Cortex M0, which is a **v6m** cpu.

(Nothing currently uses this file, as far as I can tell, and I don't know if there even exists an mpu_armv6.h)

lurch, 2022-09-17:
> I don't know if there even exists an mpu_armv6.h

Even the [upstream CMSIS](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/Core/Include) "only" has `mpu_armv7.h` and `mpu_armv8.h`.

But there **does** appear to be a bunch of MPU defines inside [core_cm0plus.h](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/cmsis/stub/CMSIS/Core/Include/core_cm0plus.h#L522) ?

kilograham, 2022-10-21:
We just package what they have; keeping this around though as we look at more CMSIS integration