raspberrypi/pico-sdk #1029: Linker warning when compiling with GCC 12

ooxi, 2022-09-19:
When compiling a [minimal example](https://github.com/ooxi/fedora-36-gcc-12-pico-sdk-linker-warning) with GCC 12.2 I'm getting the following linker warning:

```
[1/5] Performing build step for 'ELF2UF2Build'
ninja: no work to do.
[5/5] Linking CXX executable hello_world.elf
/usr/lib/gcc/arm-none-eabi/12.2.0/../../../../arm-none-eabi/bin/ld: warning: hello_world.elf has a LOAD segment with RWX permissions
/usr/lib/gcc/arm-none-eabi/12.2.0/../../../../arm-none-eabi/bin/ld: warning: hello_world.elf has a LOAD segment with RWX permissions
```

This is confusing for novices. Unfortunately my linker script foo is not sufficient enough to propose a fix.

GCC 12.2 is shipped by Fedora 36, the current stable release. You can see the linker warning with full compile output in [my GitHub Actions build](https://github.com/ooxi/fedora-36-gcc-12-pico-sdk-linker-warning/actions/runs/3084298252/jobs/4986283377#step:4:133)

aswild, 2022-09-29:
This is technically a new warning in Binutils 2.39 rather than GCC, but distros tend to update Binutils and GCC together.

The new warnings can be disabled by linking with `-Wl,--no-warn-rwx-segments`, which can be done in cmake with

```cmake
# in pico-sdk/src/rp2_common/pico_standard_link/CMakeLists.txt
target_link_options(pico_standard_link INTERFACE "LINKER:--no-warn-rwx-segments")

# or for a specific executable in a local CMakeLists.txt
target_link_options(hello_world PRIVATE "LINKER:--no-warn-rwx-segments")
```

I presume that the new flag isn't available in older versions of GNU ld so I don't know the best way for pico-sdk to fix this for everyone without doing some sort of complex linker version checking, since cmake doesn't have an equivalent of CMAKE_C_COMPILER_VERSION for the linker itself.

I tried removing the 'x' from 'rwx' in the memory sections of `mmap_default.ld` but that didn't seem to have any effect, the first two program segments are still rwx according to readelf and the warning still fires. Possibly related to use of `volatile`, c.f. https://stackoverflow.com/questions/73429929/gnu-linker-elf-has-a-load-segment-with-rwx-permissions-embedded-arm-project

drankinatty, 2022-10-16:
Confirmed with mingw-w64-x86_64-arm-none-eabi-binutils 2.39-1. This occurs when mingw64-arm-none-eabi remains at version 10.1.0 (on windows). The global fix above in `../pico_standard_link/CMakeLists.txt` works like a champ.

kilograham, 2022-10-21:
see https://www.redhat.com/en/blog/linkers-warnings-about-executable-stacks-and-segments

peterharperuk, 2022-11-23:
I don't know much about this stuff, but, just using the blink example, you can see the problem...

readelf -lW blink.elf

Elf file type is EXEC (Executable file)
Entry point 0x100001e9
There are 3 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x001000 0x10000000 0x10000000 0x01c1c 0x01c1c RWE 0x1000
  LOAD           0x0030c0 0x200000c0 0x10001c1c 0x0016c 0x0016c RWE 0x1000
  LOAD           0x000230 0x20000230 0x20000230 0x00000 0x003c8 RW  0x1000

 Section to Segment mapping:
  Segment Sections...
   00     .boot2 .text .rodata .binary_info 
   01     .data 
   02     .bss 

I think the first is writeable due to binary_info. The __bi_decl macro defines a writable pointer, so quite easy to fix. 

Change
struct _binary_info_core * name = bi
to
struct _binary_info_core * const name = bi

kilograham, 2023-01-17:
merged into develop

PR #1114 (2023-01-17): Disable linker warning
In arm-gnu-toolchain-12.2 we see this warning, that's not relevant to pico. Disable it.

warning: blink.elf has a LOAD segment with RWX permissions

Fixes #1029
commit 419eb739 (2023-01-17): Disable linker warning (#1114)

In arm-gnu-toolchain-12.2 we see this warning, that's not relevant to
pico. Disable it.

warning: blink.elf has a LOAD segment with RWX permissions

Fixes #1029
closed 2023-01-17 by kilograham