raspberrypi/pico-sdk #1030: incorrect link in getting-started-with-pico.pdf

greggjaskiewicz, 2022-09-19:
in 'installing on macOS' section, this is incorrect:
-  brew tap ArMbed/homebrew-formulae

Should be:
- brew tap ARMmbed/homebrew-formulae

lurch, 2022-09-19:
https://github.com/ARMmbed/homebrew-formulae says:
```
brew tap ArmMbed/homebrew-formulae
brew install arm-none-eabi-gcc
```

and section 9.1.1. of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf also says:
```
$ brew install cmake
$ brew tap ArmMbed/homebrew-formulae
$ brew install arm-none-eabi-gcc
```

??

greggjaskiewicz, 2022-09-21:
try it on a mac and see if it works..

lurch, 2022-09-21:
```
andrew@macmini-i3 ~ % brew tap ArmMbed/homebrew-formulae                
andrew@macmini-i3 ~ % brew install arm-none-eabi-gcc    
Warning: armmbed/formulae/arm-none-eabi-gcc 10.3-2021.07 is already installed and up-to-date.
To reinstall 10.3-2021.07, run:
  brew reinstall arm-none-eabi-gcc
```

I see in your first comment you have `ArMbed` - did you accidentally type in too few `m`s?

aallan, 2022-09-21:
I copy and pasted from the getting started guide yesterday to do a fresh Mac install and it works as written.

greggjaskiewicz, 2022-09-23:
well, it didn't work for me on my M1Pro .. ¯\_(ツ)_/¯ I have copied and pasted it from the PDF. 
The fixed link works fine, hence this issue. 
I think more people need to test this

lurch, 2022-09-23:
Given that our getting-started guide matches what https://github.com/ARMmbed/homebrew-formulae/blob/master/README.md says, perhaps you need to open an issue in that upstream repo?

MrGreensWorkshop, 2022-10-11:
This is from one of my install logs. I can confirm that, there is no problem here.

```shell
$ brew tap ArmMbed/homebrew-formulae
==> Tapping armmbed/formulae
Cloning into '/usr/local/Homebrew/Library/Taps/armmbed/homebrew-formulae'...
remote: Enumerating objects: 69, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 69 (delta 4), reused 4 (delta 2), pack-reused 60
Receiving objects: 100% (69/69), 15.08 KiB | 5.03 MiB/s, done.
Resolving deltas: 100% (31/31), done.
Tapped 1 formula (14 files, 33.5KB).

$ brew install arm-none-eabi-gcc               
==> Downloading https://developer.arm.com/-/media/Files/downloads/gnu-rm/10.3-2021.07/gcc-arm-none-eabi-10.3-2021.07-mac-10.14.
==> Downloading from https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu-rm/10.3-2021.07/gcc-arm-none-eabi-10.3
######################################################################## 100.0%
==> Installing arm-none-eabi-gcc from armmbed/formulae
...
```