raspberrypi/pico-sdk #1031: Add lwip/TinyUSB docs to the SDK docs

kilograham, 2022-09-22:
We now include Doxygen from `cyw43_driver`; we should try to add TinyUSB and lwip also (though given how fickle Doxygen is, i don't know whether we can do this directly as part of our main doxygen pass)

kilograham, 2023-01-31:
actually we should just link to their documentation site since we are HTML anyway

aallan, 2023-02-01:
Just an FYI. We're in the process of integrating the Pico SDK API directly into the documentation site, see https://github.com/raspberrypi/documentation/pull/2743 for details, and will be spinning down the existing micro-site.

lurch, 2023-02-01:
...although we'll want to keep the Doxygen build "working", as that's used by the PR Alasdair mentioned, and we'll also be keeping the "Appendix E: Building the SDK API documentation" in https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

kilograham, 2023-02-01:
yeah what i meant was that rather than trying to batch include the documentation somehow as part of our tree, we'll put links to their documentation sites in the relevant places in our docs.

aallan, 2023-02-01:
Yup. Got it!

A taster of what things should look like at the top level.

[image]

kilograham, 2023-02-05:
fixed by #1220

kilograham, 2023-02-06:
merged into develop

closed 2023-02-06 by kilograham