raspberrypi/pico-sdk #1042: Which depot to request a new PIO assembly pseudoinstruction?

henrygab, 2022-10-01:


NollKollTroll, 2024-09-29:
This would have saved me some time today, trying to come up with a dec X/Y solution. Would be really handy to not be bothered with a unique label that can potentially be wrong.

visenri, 2025-03-09:
I've also been thinking recently about a new pseudo instruction like this.
It could be handy, it makes the code more readable and less error prone.

henrygab, 2025-05-25:
This fully backward-compatible change (only changes sources, no new emitted instructions) is still a feature worth adding....

henrygab, 2025-05-26:
I admit I'm having trouble visualizing a well-formed program that, with adding the `dec {x | y}` pseudo-instruction, would fail to compile.  For example, I think a label requires `:`?   You seem more experienced with grammar parsing ... 

Could you provide simple example(s) that would cause trouble?  (They'd also be useful for tests, since they'd be important edge cases.)

henrygab, 2025-05-28:
The current compiler might choose to emit an error for current keywords, when they are used as a label.

<hr/>
<details><summary>At first glance, that appears to be a design choice, and not a hard requirement.</summary>

Here's what each line must parse as:

https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/tools/pioasm/parser.yy#L176-L185

Of interest are lines 179-181, which allow a `label_decl`, a `label_decl` followed by an `instruction`, or just an `instruction`.  The `label_decl` is defined to be any `symbol_def`, followed by `COLON`.

https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/tools/pioasm/parser.yy#L191-L192

Meanwhile, each potential `instruction` starts with a `base_instruction`:

https://github.com/raspberrypi/pico-sdk/blob/ee68c78d0afae2b69c03ae1a72bf5cc267a2d94c/tools/pioasm/parser.yy#L251-L256

And ... none of the valid `base_instruction` overlap with `symbol_def` **_because_** `symbol_def` requires a trailing colon.

And ... once a line is matching `jmp`, the `expression` would match `ID` (a deferred reference to a label).

</details><hr/>

Thus, the context of the current parsing makes it clear whether `dec` is a label or instruction.  Put another way, it appears that, if the line starts with `dec`, whether that `dec` is a label or an instruction depends on whether it is followed by `COLON` (aka `:`).  

<details><summary>Example using `dec` as both label and instruction</summary><P/>

```
.program test2
    jmp x--, dec
dec:
    nop
    dec x
    nop
```

If I manually parse that, it seems to parse without error?

</details>

Am I missing a case where there would be ambiguity, or mandatory failure?

kilograham, 2025-05-28:
note also that other things like CircuitPython support parsing the `.pio` format

henrygab, 2025-07-27:
Still hopeful that useful pseudo-instructions such as `dec` can be added.