raspberrypi/pico-sdk #1043: SDK is sensitive to special characters (spaces and parens) in the source directory name

Botje, 2022-10-03:
This bug report is a follow-up on [this StackOverflow question](https://stackoverflow.com/questions/73932381/bin-sh-1-syntax-error-unexpected-error-while-building-code-for-raspb).
The user had a project that mysteriously stopped working after taking a copy.
Crucially, the copy ended up as `serial_pico (copy)`, which fails with errors: (see SO question)
```
Scanning dependencies of target bs2_default
[  1%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  2%] Linking ASM executable bs2_default.elf
/bin/sh: 1: Syntax error: "(" unexpected
make[2]: *** [pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/build.make:98: pico-sdk/src/rp2_common/boot_stage2/bs2_default.elf] Error 2
```

Steps to reproduce:
1. Create a directory with spaces and parentheses in the name
2. Create a minimal project based on the README of this project, including the hello_world.c and pico_sdk_import.cmake.
3. Run CMake and try to build in verbose mode

On my system, the failing command is:
```
/tmp/arm-gnu-toolchain-11.3.rel1-darwin-x86_64-arm-none-eabi/bin/arm-none-eabi-gcc -mcpu=cortex-m0plus -mthumb -O3 -DNDEBUG -Wl,--build-id=none --specs=nosys.specs -nostartfiles "-Wl,--script=/tmp/test checkout (copy)/build/_deps/pico_sdk-src/src/rp2_common/boot_stage2/boot_stage2.ld" -Wl,-Map=bs2_default.elf.map CMakeFiles/bs2_default.dir/compile_time_choice.S.obj -o bs2_default.elf 
cd "/tmp/test checkout (copy)/build/pico-sdk/src/rp2_common/boot_stage2" && /tmp/arm-gnu-toolchain-11.3.rel1-darwin-x86_64-arm-none-eabi/bin/arm-none-eabi-objdump -h /tmp/test\ checkout\ (copy)/build/pico-sdk/src/rp2_common/boot_stage2/bs2_default.elf >bs2_default.dis
/bin/sh: -c: line 0: syntax error near unexpected token `('
/bin/sh: -c: line 0: `cd "/tmp/test checkout (copy)/build/pico-sdk/src/rp2_common/boot_stage2" && /tmp/arm-gnu-toolchain-11.3.rel1-darwin-x86_64-arm-none-eabi/bin/arm-none-eabi-objdump -h /tmp/test\ checkout\ (copy)/build/pico-sdk/src/rp2_common/boot_stage2/bs2_default.elf >bs2_default.dis'
```

The cause of this error seems to be src/rp2_common.cmake on lines 15 and 16. Adding the `VERBATIM` option after the `COMMAND` stanzas and adding a space after the `>` and `>>` respectively allows the build to proceed a bit further, after which it complains about a similar `objcopy` call.

peterharperuk, 2022-12-05:
It seems that adding the VERBATIM keyword to every add_custom_command fixes this.

>               If  VERBATIM  is  given then all arguments to the commands will be escaped properly
>               for the build tool so that the invoked command receives  each  argument  unchanged.
>               Note that one level of escapes is still used by the CMake language processor before
>               add_custom_target even sees the arguments. Use of VERBATIM  is  recommended  as  it
>               enables  correct  behavior.  When  VERBATIM  is  not given the behavior is platform
>               specific because there is no protection of tool-specific special characters.
>

kilograham, 2023-01-07:
merged into develop

PR #1136 (2022-12-06): Fix build issues with spaces and parentheses
Add VERBATIM to add_custom_command

Fixes #1043
commit 0b526036 (2022-12-06): Fix build issues with spaces and parentheses (#1136)

Add VERBATIM to add_custom_command

Fixes #1043
closed 2023-01-07 by kilograham