raspberrypi/pico-sdk #1044: Switch from tinyusb to tinyusb-src

sqrtroot, 2022-10-03:
Since the tiny usb repo also includes lots of sdks for other modules for their examples. If you do a `git submodule update --init --recursive` you'll get tons of sdk's you don't need. The maker of tinyusb has a seperate repo: tinyusb-src with only the src code of the project.

We should consider switching the submodule to that repository with the correct commit:
https://github.com/hathach/tinyusb_src/tree/1f1e9b3412d3f0abfba90376c8166b0c4e138115

lurch, 2022-10-03:
> Since the tiny usb repo also includes lots of sdks for other modules for their examples. If you do a `git submodule update --init --recursive` you'll get tons of sdk's you don't need.

Yup, and that's why we don't suggest doing a recursive clone in our [getting started documentation](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf).

kilograham, 2022-10-03:
agreed; this makes sense

sqrtroot, 2022-10-04:
The problem is that for some other submodules this actually might be needed (you even do it for the openocd port for the pico). The repository was created by the author to solve this problem so what would be the downside of switching?

kilograham, 2022-10-16:
@liamfraser since you are updating to the *correct*  0.14.0 commit anyway

paulhamsh, 2022-10-25:
> Yup, and that's why we don't suggest doing a recursive clone in our [getting started documentation](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf).

The TinyUSB version in the SDK seems old - it seems to be missing Pico-PIO-USB, which is odd for a Pico SDK. 
I can't work out how to get that without deleting TinyUSB, cloning from github. then doing that fun recursive submodule thing. I don't know git.

Any way to keep the TInyUSB branch up to date in this repo - or via git?

liamfraser, 2023-01-31:
Do this as a separate submodule

liamfraser, 2023-02-01:
tinyusb-src does not pull in pio usb, so we would have to pull that in somehow

kilograham, 2023-02-06:
deferred for now

shreeve, 2023-11-28:
@kilograham, @liamfraser - TinyUSB version 1.6.0 was just released. Would it be possible to update and also fold in the concept in this issue at the same time?

liamfraser, 2024-07-17:
This issue is outdated now, as tinyusb has removed submodules from the repo in favour of a custom dependency management tool

peterharperuk, 2024-08-09:
Closing as no longer relevant?