raspberrypi/pico-sdk #1046: Unexpected (CMake) behavior with PICO_BOARD variable.

tjko, 2022-10-04:
When changing PICO_BOARD variable between subsequence "cmake" invocations, there seems to be unexpected (wrong?) behavior when it comes to setting/unsetting PICO_CYW43_SUPPORTED variable.

At first this seems to work fine, passing "-DPICO_BOARD=pico_w" to cmake causes PICO_CYW43_SUPPORTED to get set.
However, if re-run cmake and specify "-DPICO_BOARD=pico", this does not unset the PICO_CYW43_SUPPORTED. And this in turn leads to CYW43 stuff being still included in the build....(which seems "wrong").

Following patch to .cmake files would seem to correct this behavior:
```
diff --git a/src/boards/generic_board.cmake b/src/boards/generic_board.cmake
index 54b43d1..188cea3 100644
--- a/src/boards/generic_board.cmake
+++ b/src/boards/generic_board.cmake
@@ -21,3 +21,5 @@ else()
     string(CONCAT msg ${msg} "   Looked for ${PICO_BOARD}.cmake in ${DIRS} (additional paths specified by PICO_BOARD_CMAKE_DIRS)")
     message(FATAL_ERROR ${msg})
 endif()
+
+set(PICO_CYW43_SUPPORTED "0" CACHE INTERNAL "Exclude support for PICO_CYW43")
diff --git a/src/boards/pico_w.cmake b/src/boards/pico_w.cmake
index 15b901d..1191731 100644
--- a/src/boards/pico_w.cmake
+++ b/src/boards/pico_w.cmake
@@ -1,2 +1,2 @@
-set(PICO_CYW43_SUPPORTED "1" CACHE INTERNAL "Try to add support for PICO_CYW43")
 include(${CMAKE_CURRENT_LIST_DIR}/generic_board.cmake)
+set(PICO_CYW43_SUPPORTED "1" CACHE INTERNAL "Try to add support for PICO_CYW43")
```

peterharperuk, 2022-10-05:
Looks good to me. Do you want to make a pull request?

kilograham, 2022-10-21:
merged into develop

closed 2022-10-21 by kilograham