raspberrypi/pico-sdk #1048: Does the PLL "tip" in the datasheet really warrant an assertion failure in pll_init

d-c-d, 2022-10-04:
Does this assert in pll_init (pll.c line 34) warrant an assertion failure?

```
    // post_div1 should be >= post_div2
    // from appnote page 11
    // postdiv1 is designed to operate with a higher input frequency
    // than postdiv2
    assert(post_div2 <= post_div1);
```
    
I was unable to find the referenced appnote page 11
and all that is mentioned in the [datasheet](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf)
was this tip.  A preference does not imply an assertion failure in my opionion.

> TIP
> When two different values are required for POSTDIV1 and POSTDIV2, it’s preferable to assign the higher value to POSTDIV1, for lower power consumption.

peterharperuk, 2022-10-05:
Duplicate? https://github.com/raspberrypi/pico-sdk/issues/1047

d-c-d, 2022-10-07:
sorry - for sure - wasn't on purpose - maybe my browser resubmited it for me :-(
will try to delete

closed 2022-10-07 by d-c-d