raspberrypi/pico-sdk #1051: Pico w "cyw43_arch_init" asserts in irq.c line 124 when run on core1

inthestillofthenight, 2022-10-06:
Requested to raise a defect on this issue, I have a simple network project written in C for the Pico w that works fine on core 0 but when moved to core 1 fails when calling cyw43_arch_init. It asserts in irq.c line 124 when run on core1. I can see it calls low_prio_irq_init(...) which calls irq_set_exclusive_handler(...) which is where it asserts.
If I put the "cyw43_arch_init" in core 0 and the other code in core 1 it init's ok, receives data but does not send data.

Cutdown code attached.
Built with VSCode running on Win11 PC with Pico used for SWD and Pico sdk 1.4
when run the output is:
It's all gone well on core 0!
Its all gone well on core 1
assertion "current == __unhandled_user_irq || current == handler" failed: file "C:\pico\pico-sdk\src\rp2_common\hardware_irq\irq.c", line 124, function: irq_set_exclusive_handler

[PicoW-Bug report network assert on core1.zip](https://github.com/raspberrypi/pico-sdk/files/9726819/PicoW-Bug.report.network.assert.on.core1.zip)

peterharperuk, 2022-10-06:
If I removed these lines, your code worked for me.

#pico_enable_stdio_usb(${PROJECT_NAME} 1)
#pico_enable_stdio_uart(${PROJECT_NAME} 1)

I got the assertion with these lines included. Which is a bit odd.

kilograham, 2022-10-06:
that is expected; the problem is an interaction with stdio_usb (although that is not the bug)

closed 2023-01-24 by kilograham