raspberrypi/pico-sdk #1054: Pico W cyw43_arch_wifi_connect_timeout_ms gives up before timeout

MrGreensWorkshop, 2022-10-11:
### Problem
Pico W cyw43_arch_wifi_connect_timeout_ms gives up before timeout.

### Expected behavior
When cyw43_arch_wifi_connect_timeout_ms is called, it is expected to connect to wifi within the given timeout.

### Steps to reproduce:
1. Make sure your AP is not turned on. (you can use smartphone hotspot AP too)
2. Plug Raspberry Pi Pico W to usb.
3. Turn your AP on.

### Test code
I created [this wifi check project](https://github.com/MrGreensWorkshop/RasPiPicoSDK_wifi_con_test) for testing its behavior.

if AP is there there at the beginning, you get connected within 10 secs.

### Log
```
connect status: joining
connect status: no ip
connect status: link up
connected
```

if AP isn't there at the beginning you get failed within 10 secs .
```
connect status: joining
connect status: network fail
failed to connect
```

### Conclusion
This is how it's called. (I'm sure you already know it)

https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_cyw43_arch/cyw43_arch.c

// cyw43_arch_wifi_connect_timeout_ms -> cyw43_arch_wifi_connect_until -> 
// {
//   cyw43_arch_wifi_connect_async -> called once
//   checks link status with cyw43_tcpip_link_status and cyw43_arch_poll
// }

I think the way its checking connection is wrong. 

Since link fails, cyw43_arch_wifi_connect_async must be called again periodically.

this could be related to #917 but I'm not sure.

MrGreensWorkshop, 2022-10-20:
I think suggested error codes for cyw43_arch_wifi_connect_timeout_ms misleading.

[Documentation](https://raspberrypi.github.io/pico-sdk-doxygen/group__pico__cyw43__arch.html#gae1237b71f7793ba603a4b39da37c7a71) suggests, See also -> **pico_error_codes**
I think these are the right ones in [cyw43.h](https://github.com/georgerobotics/cyw43-driver/blob/195dfcc10bb6f379e3dea45147590db2203d3c7b/src/cyw43.h#L80)

#define CYW43_LINK_DOWN         (0)     ///< link is down
#define CYW43_LINK_JOIN         (1)     ///< Connected to wifi
#define CYW43_LINK_NOIP         (2)     ///< Connected to wifi, but no IP address
#define CYW43_LINK_UP           (3)     ///< Connect to wifi with an IP address
#define CYW43_LINK_FAIL         (-1)    ///< Connection failed
#define CYW43_LINK_NONET        (-2)    ///< No matching SSID found (could be out of range, or down)
#define CYW43_LINK_BADAUTH      (-3)    ///< Authenticatation failure

MrGreensWorkshop, 2022-10-20:
I think suggested error codes for **cyw43_arch_wifi_connect_timeout_ms** is misleading.

[Documentation](https://raspberrypi.github.io/pico-sdk-doxygen/group__pico__cyw43__arch.html#gae1237b71f7793ba603a4b39da37c7a71) suggests, See also -> [**pico_error_codes**](https://raspberrypi.github.io/pico-sdk-doxygen/group__pico__base.html#gaddcb20bc9a61ceff479aa87369a862aa)

```C
enum pico_error_codes {
  PICO_OK = 0 , 
  PICO_ERROR_NONE = 0 , 
  PICO_ERROR_TIMEOUT = -1 , 
  PICO_ERROR_GENERIC = -2 ,
  PICO_ERROR_NO_DATA = -3 , 
  PICO_ERROR_NOT_PERMITTED = -4 , 
  PICO_ERROR_INVALID_ARG = -5 , 
  PICO_ERROR_IO = -6
}
```

I think these are the right ones from [cyw43.h](https://github.com/georgerobotics/cyw43-driver/blob/195dfcc10bb6f379e3dea45147590db2203d3c7b/src/cyw43.h#L80)
I mean, return value of **cyw43_arch_wifi_connect_until** can be **PICO_OK**, **PICO_ERROR_TIMEOUT** or one of these below.

```C
#define CYW43_LINK_DOWN         (0)     ///< link is down
#define CYW43_LINK_JOIN         (1)     ///< Connected to wifi
#define CYW43_LINK_NOIP         (2)     ///< Connected to wifi, but no IP address
#define CYW43_LINK_UP           (3)     ///< Connect to wifi with an IP address
#define CYW43_LINK_FAIL         (-1)    ///< Connection failed
#define CYW43_LINK_NONET        (-2)    ///< No matching SSID found (could be out of range, or down)
#define CYW43_LINK_BADAUTH      (-3)    ///< Authenticatation failure
```

in this case it is **CYW43_LINK_NONET**

peterharperuk, 2023-01-23:
> I mean, return value of [cyw43_arch_wifi_connect_until](https://github.com/raspberrypi/pico-sdk/blob/2e6142b15b8a75c1227dd3edbe839193b2bf9041/src/rp2_common/pico_cyw43_arch/cyw43_arch.c#L73) can be PICO_OK, PICO_ERROR_TIMEOUT or one of these below.

or CYW43_EPERM  or CYW43_EIO

peterharperuk, 2023-01-23:
After making this change, cyw43_arch_wifi_connect_blocking could potentially never return. I wonder if that's a problem

kilograham, 2023-01-23:
^ note we should either return CYW43_ errors or PICO_ erros, but not a mixture of both

MrGreensWorkshop, 2023-01-24:
@peterharperuk Thanks for the fix. I didn't test it, but it looks fine.

The only place that PICO_errors used is, cyw43_arch_wifi_connect_until (PICO_ERROR_TIMEOUT) in `cyw43_arch.c/h` I agree with @kilograham, that mixing things up can be confusing. I prefer to have a time-out error in CYW43_ errors to avoid confusion. 

Also, `see pico_error_codes` in `cyw43_arch.h` file needs to be fixed for other functions too.

I believe documentation is generated with doxygen, it will be fixed automatically.

peterharperuk, 2023-01-24:
> The only place that PICO_errors used is, cyw43_arch_wifi_connect_until

cyw43_arch_wifi_connect_until calls cyw43_arch_wifi_connect_async which returns pico_error_codes.

> there is no way to tell where those are coming from

CYW43_EPERM or CYW43_EIO are defines used in the cyw43-driver that are set to pico_error_codes. The cyw43-driver is used by other platforms.

peterharperuk, 2023-01-25:
I had to create a new PR, my github skillz failing me.

kilograham, 2023-01-27:
merged into develop

commit b979395c (2023-01-26): Keep retrying if cyw43_wifi_join fails (#1195)

Currently, if the network can't be found we fail almost immediately.
Also fix error codes.

Fixes #1054
closed 2023-01-27 by kilograham