raspberrypi/pico-sdk #1055: SWD no longer works after a verify fail during flashing.

jondurrant, 2022-10-11:
I have a strange one. A SWD load of a c program failed to verify. the device is no longer detected via openocd to flash. I get the error "DAP init failed". The device can still be flashed via bootsel. Any ideas how to reset SWD?

I've done this to three different RP2040 devices; Pico, Pico W and Rp2040 tiny, all using the same bad code obviously. So something in the code can cause this. I've also checked SWD using both picoprobe and a RPI bitbang for SWD.

I have tried completely wiping flash using flash_nuke but makes no difference.

Anyone got any ideas? Is there any persistent register for SWD config that could have been corrupted?

Thanks

Jon

jondurrant, 2022-10-11:
Just an update to what the code might of incorrectly done to cause this. I had a loop initialising a set of GPIO outputs where the stop condition was corrupt. Therefore I have initialised and sent highs and lows to GPIO pads well beyond GP27.

Wren6991, 2022-10-11:
> Therefore I have initialised and sent highs and lows to GPIO pads well beyond GP27.

Pad register indices 30/31 (beyond the GPIOs) have some (rather useless) pad controls for the SWCLK/SWDIO pads. If bad values were written to those then you could have disconnected the internal SWD bus from the pads. (Another way to achieve this is with the SWD internal bitbang registers in `syscfg`.)

So, that would explain the death of SWD, but it's then a little surprising that flash nuke plus a full system reset (either the RUN pin or a power cycle) would not restore it. There is no persistent storage on RP2040. In fact I would have expected just a reset into the USB bootloader to be sufficient to bring back SWD.

kilograham, 2022-10-11:
`openocd` is not above getting itself into a bad state; did you also reset openocd (or are your running it fresh every time)?

jondurrant, 2022-10-11:
I am running openocd fresh everytime. Have also run openocd using PI GPIO and a Pico with Picoprobe. Also rebooted the PI.

Really looks like something is preserved on the Pico, which doesn't seem possible. As @Wren6991 said I want to look at the syscfg register and see if different on the non working Pico's compared with the working units.

jondurrant, 2022-10-13:
I did a check on syscfg_hw->dbgforce register on a working a non-working Pico. Both report 0x77. Which I guess makes sense as we all agree nothing should persisting in SWD configuration between power cycles.

So what else could I have damaged through software on the Pico? A device that advertises it is not possible to damage the device through software. Yet I seem to have killed SWD on four devices. But have it working on other frest devices.

jondurrant, 2022-10-13:
Yes I have. I can load code on all the time through BOOTSEL including flash_nuke.uf2.

aallan, 2022-10-13:
So if you now unplug it, and then plug it back in but **without** pressing BOOTSEL again, it should appear as a USB device automatically. Does it do so? If it doesn't then the flash hasn't actually been properly wiped.

jondurrant, 2022-10-13:
Yes it does.

jondurrant, 2022-10-13:
If I've got good running code on the device, and it has just been through a power cycle then what could prevent SWD functioning?

aallan, 2022-10-13:
As far as we're concerned, nothing. 

Would it be possible for you to provide us with a copy of the source so we can try and replicate the problem? If it is the code, we should see the same results, and then we can poke at it.

jondurrant, 2022-10-13:
I will try. I fixed the code, but I do know what I did wrong. So I will try to reproduce a copy that will cause this. As I have managed to achieve it on two Pico W, a Pico and a RP2040 Tiny, I'm fairly certain I can reproduce.

jondurrant, 2022-10-13:
Closing issue as after rechecking it is only one of my Pico, the one I mainly work with, that won't recover SWD. So I think we should put it down to a hardware failure that happened in parallel.

closed 2022-10-13 by jondurrant