raspberrypi/pico-sdk #1059: multicore_lockout_start_blocking assertion failure v1.4.0 and v1.4.1

RaresAil, 2022-10-12:
Similar with an old closed issue: #687 

```
assertion "us_since_boot <= INT64_MAX" failed: file
```

The repo where I have the code is here, is a pretty complex one
- core 0 is handling wifi and tcp socket server
- core 1 is handling some service specific data, in the case where I tried the functions is for an adjustable desk

Repo: https://github.com/RaresAil/pico-w-template/tree/feat/flash-saving-data

Called from the main thread of core 1:

```cpp
void save_flash_data() {
  uint8_t* bytes = (uint8_t*) &flash_data;
  int data_size = sizeof(flash_data);

  printf("[Flash] Programming flash (Size: %d)\n", data_size);

  printf("[Flash] Requesting core 0 to pause\n");
  multicore_lockout_start_blocking(); // ERROR HERE
  printf("[Flash] Locking core 1\n");

  uint32_t interrupts = save_and_disable_interrupts();
  flash_range_erase(FLASH_TARGET_OFFSET, FLASH_SECTOR_SIZE);
  flash_range_program(FLASH_TARGET_OFFSET, bytes, FLASH_PAGE_SIZE);
  restore_interrupts(interrupts);

  printf("[Flash] Unlocking core 1\n");
  multicore_lockout_end_blocking();
  printf("[Flash] Releasing core 0\n");
  printf("[Flash] Done.\n");
}
```

At the beginning of core 0 in main
```cpp
  multicore_launch_core1(core1_entry);
  multicore_lockout_victim_init();
  ```

kilograham, 2022-10-16:
are you definitely sure you tried this with 1.4.1?

RaresAil, 2022-10-17:
> are you definitely sure you tried this with 1.4.1?

I used the develop branch when building and tried but the same issue

darrena092, 2022-11-27:
Hitting the same issue here on 1.4.0 FWIW. Will try on develop tomorrow at some point.

closed 2022-12-30 by RaresAil