raspberrypi/pico-sdk #1060: Pico W uint64_t is represented wronglyt by printf() when handled by irq callback

extremq, 2022-10-12:
### The issue
Any number in `uint64_t` form is not correctly represented by `printf` using the `%llu` format.
### Example code
Uncomment any of the `printf()` both in main and callback, but at most 4-5 printfs otherwise it's too much to print for the Pico W.
```c
/*
 * TEST for checking the printf representation.
 * Connect gpio0 at any kind of input.
 * I recommend using a button connected to the 3.3v output
 */
#include <stdio.h>

#include "pico/stdlib.h"
#include "hardware/gpio.h"

uint64_t constant = 0x0;

// Copied from sdk
static uint64_t get_time(void) {
    uint32_t lo = timer_hw -> timelr;
    uint32_t hi = timer_hw -> timehr;

    return ((uint64_t) hi << 32u) | lo;
}

// lo put in high instead of hi
static uint64_t get_time_inverse(void) {
    uint32_t lo = timer_hw -> timelr;
    uint32_t hi = timer_hw -> timehr;

    return ((uint64_t) lo << 32u) | hi;
}

// Called when on edge rise at pin 0
void gpio_callback(uint gpio, uint32_t events) {
    printf(" irq time 32 %lu\n", time_us_32());
    printf(" irq time 64 %llu\n", time_us_64());
    printf(" irq timer_hw_lr registers  0x%016lX\n", timer_hw -> timelr);
    printf(" irq timer_hw_hr registers  0x%016lX\n", timer_hw -> timehr);
    printf(" irq timer_hw registers     0x%016llX\n", get_time());
//    printf(" irq timer_hw inv registers 0x%016llX\n", get_time_inverse());
//    printf(" irq test cnst as uint64_t  0x%016llX\n", constant);
}

int main() {
    stdio_init_all();

    // Enable irq when toggling to 1 on gpio 0
    gpio_init(0);
    gpio_set_irq_enabled_with_callback(0, GPIO_IRQ_EDGE_RISE, true, &gpio_callback);

    while(1) {
        // Display time in the while loop - should be correct
//        printf("[loop] Attempting to print %llu\n", constant);
        printf("loop time 32 %lu\n", time_us_32());
        printf("loop time 64 %llu\n", time_us_64());
        printf("loop timer_hw_lr registers  0x%016lX\n", timer_hw -> timelr);
        printf("loop timer_hw_hr registers  0x%016lX\n", timer_hw -> timehr);
        printf("loop timer_hw registers     0x%016llX\n", get_time());
//        printf("loop timer_hw inv registers 0x%016llX\n", get_time_inverse());
//        printf("loop test cnst as uint64_t  0x%016llX\n", constant);

        sleep_ms(1000);
        ++constant;
    }
    return 0;
}
```

### Example output
```
loop time 32 9005628  // correct                                                          
loop time 64 9005698  // correct                                                          
loop timer_hw_lr registers  0x0000000000896AE0 // lower 32 bits correct                                 
loop timer_hw_hr registers  0x0000000000000000                                  
loop timer_hw registers     0x0000000000896B92                                  
 irq time 32 9592652                                                            
 irq time 64 41200418679685120 // incorrect, should be 9592720                                                   
 irq timer_hw_lr registers  0x0000000000925FD4 // lower 32 bits correct                                  
 irq timer_hw_hr registers  0x0000000000000000                                  
 irq timer_hw registers     0x0092605B00000000 // bits aren't represented correctly
 ```
### Additional info
I've tested printing a `uint64_int` variable, then adding `1` to it, and then printing it again. The number was correct, but shifted onto the upper 32 bits of the 64-bit output. This also happens using the example in `pico-examples/gpio`.
For some reason, numbers on the lower 32 bits are shifted on the higher 32 bits.

extremq, 2022-10-12:
### Additional additional info
```
[loop] Attempting to print 16
loop test cnst as uint64_t  0x0000000000000010
 irq test cnst as uint64_t  0x0000001000000008
[loop] Attempting to print 17                ^
loop test cnst as uint64_t  0x0000000000000011
 irq test cnst as uint64_t  0x0000001100000008
[loop] Attempting to print 18                ^
loop test cnst as uint64_t  0x0000000000000012
 irq test cnst as uint64_t  0x0000001200000008
[loop] Attempting to print 19                ^
loop test cnst as uint64_t  0x0000000000000013
 irq test cnst as uint64_t  0x0000001300000008
[loop] Attempting to print 20                ^
loop test cnst as uint64_t  0x0000000000000014
 irq test cnst as uint64_t  0x0000001400000008
[loop] Attempting to print 21                ^
loop test cnst as uint64_t  0x0000000000000015
 irq test cnst as uint64_t  0x0000001500000008
[loop] Attempting to print 22                ^
loop test cnst as uint64_t  0x0000000000000016
 irq test cnst as uint64_t  0x0000001600000008
[loop] Attempting to print 23                ^
loop test cnst as uint64_t  0x0000000000000017
 irq test cnst as uint64_t  0x0000001700000008
[loop] Attempting to print 24                ^
loop test cnst as uint64_t  0x0000000000000018
 irq test cnst as uint64_t  0x0000001800000008
 ```
 When printing constants, not only they get shifted but they also have an `8` at the end.

lurch, 2022-10-12:
https://stackoverflow.com/a/9225648

kilograham, 2022-10-12:
try the `develop` branch; #971 may affect this

extremq, 2022-10-12:
> https://stackoverflow.com/a/9225648

Not sure what you meant by that, but I appreciate the interest!

> try the `develop` branch; #971 may affect this

Will try and update you on this.

extremq, 2022-10-12:
```
 irq time 32 17814766                                                           
 irq time 64 17814825                                                           
 irq timer_hw_lr registers  0x00000000010FD556                                  
 irq timer_hw_hr registers  0x0000000000000000                                  
 irq timer_hw registers     0x00000000010FD5D1                                  
loop time 32 18007349                                                           
loop time 64 18007412                                                           
loop timer_hw_lr registers  0x000000000112C5D0                                  
loop timer_hw_hr registers  0x0000000000000000                                  
loop timer_hw registers     0x000000000112C6AA
```
Works as expected. Thanks for the help!

closed 2022-10-12 by extremq