raspberrypi/pico-sdk #1061: pico-sdk documentation issues

pingufreak, 2022-10-15:
Hello together,

I'm currently going through the pico-sdk documentation and I've notice some really basic faults. For example in chaper 1.2. Anatomy of a SDK Application "int main()" is mentioned with a PICO_DEFAULT_LED_PIN macro. The main method main() must return an int. Else some strange behaviours can appear (for example the internals of the uC does wait for an int... , the usb serial output will not work, putty will crash!).

Where can I contribute to improve the documentation? If I find other issues, I would like to send updates.

Thank you!

Kind regards,

pingu

lurch, 2022-10-15:
> Where can I contribute to improve the documentation? If I find other issues, I would like to send updates.

The source of the documentation isn't public, but please report any updates or corrections by opening issues at https://github.com/raspberrypi/pico-feedback/
Thanks.

P.S. Did you read the parts where it says "It’s quite common for the main() function not to return, as is shown here." and "The return code of main() is ignored by the SDK runtime, and the default behaviour is to hang the processor on exit." ?

pingufreak, 2022-10-15:
Thank you for the fast answer! I will report there then. I've read this, but it seems that usb-serial is not working properly if main doesn't returns something.

lurch, 2022-10-15:
Could you provide more details about "usb-serial is not working properly"? Are you perhaps running into the same "problem" I described [here](https://github.com/raspberrypi/pico-examples/issues/110#issuecomment-836596266)? If so, you might want to try increasing `PICO_STDIO_USB_CONNECT_WAIT_TIMEOUT_MS` (or set it to `-1`).