raspberrypi/pico-sdk #1071: Add "-b <addr>" option to elf2uf2

esky-software, 2022-10-18:
The input file is treated as a binary blob to be packaged at the specified address.

Handy when the application expects to find data at a known location in flash.
e.g. Config data, display resources, etc.

kilograham, 2022-10-21:
Note there is a python script to do that here `https://github.com/microsoft/uf2/`

esky-software, 2022-10-21:
Heh, so there is: https://github.com/microsoft/uf2/blob/master/utils/uf2conv.md
Feel free to merge or close the PR as you see fit.

kilograham, 2022-10-21:
Yeah - it is a nice idea, but feels a bit weird in a tool called `elf2uf2`... but we will add such support if we have reason to add functionality related to making UF2s in the future

lurch, 2024-11-11:
@will-v-pi Has this been "fixed" in picotool? I guess at the very least, this issue ought to be moved over to that repo?

will-v-pi, 2024-11-11:
Yes, it’s been added - the option is `-o`. I closed the PR but missed this issue

closed 2024-11-11 by will-v-pi