raspberrypi/pico-sdk #1073: rp2040.svd file device's description is missing

juju2013, 2022-10-18:
It looks like there's an error in the [rp2040.svd](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2040/hardware_regs/rp2040.svd) file.

Per specification of [CMSIS-SVD](https://raw.githubusercontent.com/ARM-software/CMSIS/master/CMSIS/Utilities/CMSIS-SVD.xsd) (near the text "`description is a string describing the device features (e.g. memory size, peripherals, etc.)`"), the field "_description_" of element "_device_" is mandatory.

In the actual rp2040.svd, this field is missing.

lurch, 2022-10-18:
https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_device.html doesn't describe the `description` as required. Or does "Occurrence: 1..1" actually mean the same thing as "Use: required"? :thinking: 

Actually, looking at the file that you linked to:
```
        <!-- version specifies the version of the device description -->
        <xs:element name="version" type="stringType"/>
        <!-- description is a string describing the device features (e.g. memory size, peripherals, etc.) -->
        <xs:element name="description" type="stringType"/>
        <!-- V1.1: licenseText specifies the file header section to be included in any derived file -->
        <xs:element name="licenseText" type="stringType" minOccurs="0"/>
```
where does it say that `description` is mandatory? (I'm not being antagonistic, I'm just trying to work out what I've missed)

juju2013, 2022-10-19:
Not a big fan of XML, so maybe I'm in the wrong, but according to [this specification](https://www.w3.org/TR/xmlschema-0/#OccurrenceConstraints), `use` is for attributes and `min/maxOccurs` are for elements, and in the latter case, the default value is 1.

Where things become complicated: register name is declared alike while peripheral name explicitly says `minOccurs="1"` - both of them are obviously mandatory ... what's the author's exact intent is not clear for me here.

juju2013, 2022-10-22:
When passing rp2040.svd [w3c's validator](https://www.utilities-online.info/xsdvalidation), it appears that not only the description field of device is mandatory, but there's also other issues.

So I'll close this issue and open a more generic one.

closed 2022-10-22 by juju2013