raspberrypi/pico-sdk #1077: cyw43_driver only works with SM0

Lord-Lucan, 2022-10-26:
The cyw43_driver (see cyw43_bus_pio_spi.c) only supports SM0. It tries to play nice by allocating a free PIO and requesting a SM but if it gets an SM other than 0 it doesn't work.

The workaround is to make sure that you keep SM 0 free on a PIO for this driver.

In cyw43_bus_pio_spi.c (line 263) we have code like this ....
` dma_channel_configure(bus_data->dma_out, &out_config, &bus_data->pio->txf[0], tx, tx_length / 4, true);`
Which is hardwiring the DMA to pio SM 0 fifo.

Have tried to fix, but not completely working yet :(

kilograham, 2022-10-26:
I would thing just changing from `txf[0]` to `txf[bus_data->pio_sm]` should do it, no? (also for the other `txf[0]/rxf[0]`)

Lord-Lucan, 2022-10-26:
Hi, thanks for the reply. Yes tried all of that, there are other places in the code where it's hard coded as well. But it still doesn't transfer properly. So it's a bit more deep rooted. I'll take a deeper look at the weekend. At the moment I've just made sure my SM is not claiming 0.

Lord-Lucan, 2022-10-26:
I was wondering if it might have to do with the offset in the SM memory. Because I bet it's never had to share with another program.

Lord-Lucan, 2022-10-26:
Worked out what the final issue was, there needed to be a change to the fdebug flag for the SM in use, see PR.

kilograham, 2022-10-27:
merged into develop

PR #1080 (2022-10-27): Fix for cyw43 driver working with SMs other than 0
Fixes #1077

cyw43 driver is currently hard wired to use state machine (SM) 0. If a user program is allocated the same pio and SM 0 as cyw43 then the program will fail. Both cyw43 and user program will fall over each other trying to use SM 0.

This fix updates the cyw43 driver so that it doesn't assume that it has unique access to SM 0.

The cyw43 driver has a nod to playing fairly with the pio and sm but didn't quite make the required changes in cyw43_spi_transfer :)
closed 2022-10-27 by kilograham