raspberrypi/pico-sdk #1078: "cw43 warn" causing deadlock on freertos smp

p8a, 2022-10-26:
I'm running into a deadlock on raspberry pico w when using freertos smp kernel due to CYW43 WARN printing functions using vprintf - this is the stack:
`
Thread 1 received signal SIGINT, Interrupt.
0x100125c8 in vPortRecursiveLock (uxAcquire=1, pxSpinLock=0xd000013c, ulLockNum=1) at /Users/p8a/data/test/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/include/portmacro.h:192
192	                while ( __builtin_expect( !*pxSpinLock, 0 ) );
(gdb) bt
#0  0x100125c8 in vPortRecursiveLock (uxAcquire=1, pxSpinLock=0xd000013c, ulLockNum=1) at /Users/p8a/data/test/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/include/portmacro.h:192
#1  vTaskEnterCritical () at /Users/p8a/data/test/FreeRTOS-Kernel/tasks.c:5367
#2  0x1001313c in xTaskGetSchedulerState () at /Users/p8a/data/test/FreeRTOS-Kernel/tasks.c:5041
#3  0x10010804 in xEventGroupWaitBits (xEventGroup=0x2000c5b8 <ucHeap+12>, uxBitsToWaitFor=131584, xClearOnExit=xClearOnExit@entry=1, xWaitForAllBits=xWaitForAllBits@entry=0, xTicksToWait=xTicksToWait@entry=4294967295) at /Users/p8a/data/test/FreeRTOS-Kernel/event_groups.c:333
#4  0x100106a4 in vPortLockInternalSpinUnlockWithWait (pxLock=pxLock@entry=0x200003d0 <print_mutex>, ulSave=<optimized out>) at /Users/p8a/data/test/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/port.c:788
#5  0x200000e8 in mutex_enter_blocking (mtx=mtx@entry=0x200003d0 <print_mutex>) at /Users/pba/data/mine/pico/pico-sdk/src/common/pico_sync/mutex.c:44
#6  0x1000417a in stdout_serialize_begin () at /Users/pba/data/mine/pico/pico-sdk/src/rp2_common/pico_stdio/stdio.c:52
#7  0x100042a4 in __wrap_vprintf (format=0x10015ad4 "[CYW43] got unexpected packet %d\n", va=...) at /Users/pba/data/mine/pico/pico-sdk/src/rp2_common/pico_stdio/stdio.c:236
#8  0x100042ec in __wrap_printf (format=0x10015ad4 "[CYW43] got unexpected packet %d\n") at /Users/pba/data/mine/pico/pico-sdk/src/rp2_common/pico_stdio/stdio.c:265
#9  0x10006100 in cyw43_ll_process_packets (self_in=self_in@entry=0x20000764 <cyw43_state>) at /Users/pba/data/mine/pico/pico-sdk/lib/cyw43-driver/src/cyw43_ll.c:1214
#10 0x10007406 in cyw43_poll_func () at /Users/pba/data/mine/pico/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:219
#11 0x10004822 in cyw43_task (param=<optimized out>) at /Users/pba/data/mine/pico/pico-sdk/src/rp2_common/pico_cyw43_arch/cyw43_arch_freertos.c:99
#12 0x100103c8 in xPortPendSVHandler () at /Users/p8a/data/test/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/port.c:405
Backtrace stopped: previous frame identical to this frame (corrupt stack?)`

The messages that I'm seeing in the stdout are lots of:  
`
CYW43] got unexpected packet -9
[CYW43] got unexpected packet -9
[CYW43] got unexpected packet -9`

Is there a way to disable warning messages (and all other printing using vprintf for that matter) in CYW43 ?

peterharperuk, 2022-10-26:
Sounds like a duplicate of https://github.com/georgerobotics/cyw43-driver/issues/33
See also https://github.com/georgerobotics/cyw43-driver/issues/36
But the cause of the deadlock is reported here https://github.com/raspberrypi/pico-sdk/issues/1039

It might be a good idea for us to remove the warnings. I don't think there's a way to do it without modifying the source to change CYW43_WARN.

p8a, 2022-10-26:
I ended up changing the CYW43_PRINTF macro to also cover debug, info, etc:
```
#define CYW43_PRINTF(...)  (void)0
```

there seem to be other places where prints if called like `cyw43_dump_stats`

kilograham, 2023-02-06:
merged into develop

closed 2023-02-06 by kilograham