raspberrypi/pico-sdk #1079: LWIP stability issues with background and polled LWIP modes

earlephilhower, 2022-10-26:
(moved from #910) discussion

LWIP is not IRQ/multi-thread safe/re-entrant in the `NO_SYS` mode, at all.  See https://www.nongnu.org/lwip/2_0_x/pitfalls.html .

## pico_cyw43_arch LWIP IRQ problems (background mode only)

The CYW43 wrapper is calling `sys_check_timeouts` from an alarm/IRQ here: https://github.com/raspberrypi/pico-sdk/blob/2e6142b15b8a75c1227dd3edbe839193b2bf9041/src/rp2_common/pico_cyw43_arch/cyw43_arch_threadsafe_background.c#L83

99% of the time this is fine in a lightly loaded system as you're `wfe()`ing or doing non-network processing.  But under load you ~~may~~ **will** end up inside LWIP code when the IRQ fires and `sys_check_timeouts` ends up modifying LWIP state.  When that happens you **will** end up crashing or getting stuck in infinite loops as LWIP becomes inconsistent internally.

A corollary of this is that the `sys_check_timeouts` can end up calling LWIP user-level callbacks.  These callbacks will then execute under an IRQ context so things like memory operations (i.e. copying a TCP packet out to do some work) are illegal.  

## pico_cyw43_arch CYW43 Ethernet IRQ problems (background and polled mode)

Also, the CYW43 driver 
https://github.com/georgerobotics/cyw43-driver/blob/195dfcc10bb6f379e3dea45147590db2203d3c7b/src/cyw43_lwip.c#L219 is also using an LWIP `netif->input()` call from within an IRQ callback from the WiFi chip.  The same issue happens at this point because LWIP isn't re-entrant in NO_SYS.  You could be within an LWIP block and hit this line and the LWIP state will become borked.  [Note that the PBUF calls in this callback are legal but you need to implement a fix for the next bit]

## SYS_ARCH_PROTECT/UNPROTECT should be defined (background and polled mode with `MEM_LIBC_MALLOC == 0`)
This is more of a `pico-examples` thing, but the LWIP memory management routines are not re-entrant (see above).  `pbuf_xxxx` is legal to call from within an interrupt context, but it is not re-entrant.  So, AIUI, interrupts need to be disabled for safety by defining the `SYS_ARCH/(UN)PROTECT` macros to implement an interrupt disable stackso you can be sure the main app isn't doing pbuf work while the Ethernet interrupt comes in.

## Potential workaround
A potential (extreme) solution to this would be to use the linker wrap `-Wl--wrap=...` option to wrap all LWIP calls and set a semaphore so that the `sys_check_timeouts` and the `netif->input()` call can be skipped if the IRQ happens while w/in LWIP code.  My own code ended up doing this with simple wrappers and either skipping the `sys_check_timeouts` call and the `netif->input` calls (dropping the packet) while inside an LWIP call.  It's not particularly pretty but it is straightforward.

kilograham, 2022-10-26:
So to recap, yes, LWIP is known to not be re-entrant.

Because, when using *threadsafe_background*, we call from an IRQ context (holding a lock) it is imperative that other uses of LWIP (outside of the IRQ context where most lwip callbacks run) also take the lock to prevent re-entering non re-entrant code. Facilities are provided for this.

Therefore it is important to clarify which of the above problems relate to other use of LWIP without holding the lock, vs problems in the actual cyw43/lwip implementation

To the points:

#### pico_cyw43_arch LWIP IRQ problems (background mode only)

yes this is a bug (`sys_check_timeouts` should be called with the lock owned)

#### pico_cyw43_arch CYW43 Ethernet IRQ problems (background and polled mode)

I believe this is called under lock already, can you elaborate on how the problem manifests

#### SYS_ARCH_PROTECT/UNPROTECT should be defined (background and polled mode with MEM_LIBC_MALLOC == 0)

in background mode, what is calling this without holding the lock?
in polled mode, things should only be called from one context anyway!?

kilograham, 2022-10-26:
note, i'm trying to understand the problem better - I'm all for making it more automatic to have some other uses of lwip APIs take the lock, but we can't do that for everything (although perhaps we only need to do it for SYS_ARCH_PROTECT places anyway, which would be a vote for that)

earlephilhower, 2022-10-26:
> pico_cyw43_arch LWIP IRQ :  Because, when using threadsafe_background, we call from an IRQ context (holding a lock) it is imperative that other uses of LWIP (outside of the IRQ context where most lwip callbacks run) also take the lock to prevent re-entering non re-entrant code. Facilities are provided for this.

I think we're getting to the same solution slightly different ways, but yes we need to make sure the `sys_check_timeouts` is not called when the user is inside the LWIP library.  It's always perfectly fine to just skip that call...it'll have another chance later on anyway.

I assume you mean this mutex:  https://github.com/raspberrypi/pico-sdk/blob/2e6142b15b8a75c1227dd3edbe839193b2bf9041/src/rp2_common/pico_cyw43_arch/cyw43_arch_threadsafe_background.c#L68

That lock is `static`, so it's not usable by external app code, no?  Also, it's a `recursive_mutex` which allows recursion on the same core which is not what we need here (the same core is generally doing the IRQ as the main LWIP app code) because the main app could grab it, and then the IRQ could also grab it since it's the same core, no?

Maybe there's a bit in the pico-example code you could point to clarify how it should be used?  I feel like I've missed something but can't seem to find any locking on the LWIP calls in the examples I saw.

> pico_cyw43_arch CYW43 Ethernet :  I believe this is called under lock already, can you elaborate on how the problem manifests

The code's actually in the CYW43 library here
https://github.com/georgerobotics/cyw43-driver/blob/195dfcc10bb6f379e3dea45147590db2203d3c7b/src/cyw43_lwip.c#L207-L225

It's called from
https://github.com/georgerobotics/cyw43-driver/blob/195dfcc10bb6f379e3dea45147590db2203d3c7b/src/cyw43_ll.c#L1198-L1217

There is no locking around it that I can find (being a submodule, I wouldn't think it knew anything about the SDK's locks anyway).

What happens is the netif->input() tries to stuff the packet into a linked list somewhere for LWIP processing.  But if LWIP was in the middle of working on that list (i.e. handling the prior packet, or preparing a new one for transmission) the linked list gets borked and I've seen both crashes and a single element's `next` pointer point to itself, causing an infinite loop in the LWIP processing.  I'll try and see if I can find the LWIP mailing list email I found describing someone having the same issue on another processor w/IRQ-received packets.  FWIW, this happened way less frequently than the LWIP re-entrancy because of the periodic timeouts check, but it did happen under continuous load.

> SYS_ARCH_PROTECT/UNPROTECT :  in background mode, what is calling this without holding the lock?  in polled mode, things should only be called from one context anyway!?

User code isn't using this, but this is used at least by the LWIP static memory allocator (PBUF).    The issue here is that we get Ethernet packets inside an IRQ and need to use `pbuf`s to feed it into the IP stack.  So user code could have a pbuf operation (i.e. you'd do this when you complete processing of a packet) in flight in the LWIP memory manager and the Ethernet IRQ could attempt to do an allocation.  If you're not using the static allocation I *think* it can be a no-op as long as your malloc is properly re-entrant.  For Newlib this is definitely not the case (and hence the use of the static LWIP memory allocation).

>  I'm all for making it more automatic to have some other uses of lwip APIs take the lock, but we can't do that for everything

I hear you. :)  It's a pain but it is logically **possible** to wrap all the LWIP calls at the library level (see https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/lwip_wrap.cpp ) and that takes care of things once and for all.  I handled it that way because there's a ton of TCP/IP code in the Arduino core that I've borrowed from elsewhere, and even after manually blocking things out with an RAII motex, I was still finding edge cases I'd missed.  (Note I only wrapped calls I found in the code, so there are some LWIP calls that are not wrapped....but in my use case the Arduino end user is not expected to use raw LWIP calls anyway).

kilograham, 2022-10-26:
w.r.t locking, the methods are documented here:

https://github.com/raspberrypi/pico-sdk/blob/2e6142b15b8a75c1227dd3edbe839193b2bf9041/src/rp2_common/pico_cyw43_arch/include/pico/cyw43_arch.h#L261

Also read the top of the file... these are to be used if you are calling lwip from a context other than a lwip callback

---

No need to rehash `sys_check_timeouts` - as I have said, this is a bug that needs to be fixed (note, this stuff is being reworked some at the moment, so will be fixed when that lands)

earlephilhower, 2022-10-26:
Ah, did not see that document, sorry!  So, I think the first issue, as long as the timeouts callback is protected is not a problem just a user error.

I dug some more and there is checking on blocking on the Ethernet polling function:
https://github.com/georgerobotics/cyw43-driver/blob/195dfcc10bb6f379e3dea45147590db2203d3c7b/src/cyw43_ctrl.c#L208
which is defined as `#define CYW43_THREAD_LOCK_CHECK cyw43_arch_lwip_check();`

But the way it's being called it will continue the poll fcn and do the illegal `netif->input` even if the check fails (the check fcn will `panic` in debug mode).  Either option may not the best way of handling things...especially because as a user you have no idea when or control over the Ethernet background poll is happening.

In the Ethernet poll case, it seems like it would be OK to skip the poll that cycle and try again on the next loop to see if we're free, no?  Same as the sys_check_timeouts alarm will do, once patched.

kilograham, 2022-10-27:
`cyw43_poll_func()` should *only* ever be called while owning the lock, so the `assert` behaivor is correct.

It would be interesting to know why this is being called without owning the lock (which is what you are suggesting). it could be a knock on effect from the improper locking of `sys_check_timeouts` I suppose

earlephilhower, 2022-10-27:
Yes, the `low_prio_irq_handler` does grab the mutex and check the mutex count is 1 (i.e. it is the only owner) so looks like this was just a red herring on my part.  Given that, the IRQ disable in the SYS_ARCH_PROTECT is a don't care with the code in the SDK proper.

Apologies for the false alarm!

closed 2022-10-27 by earlephilhower