raspberrypi/pico-sdk #1081: Cannot drag a UF2 to RPI-RP2 on macOS Ventura

kilograham, 2022-10-26:
This results in:

[image]

It is still possible to copy UF2 files from the command line

kilograham, 2022-10-26:
Note this isn't an issue with the SDK, but this seems like as good a place as any for discussion

kilograham, 2022-10-26:
Note the error is `kPOSIXErrorENOATTR ("Attribute not found")`

This (and the fact that `cp` command doesn't have the problem) makes me suppose the issue relates to Finder trying to write and then read additional metadata (or a resource or perhaps .DS_Store) to/from the drive.

Note that the USB MSC volume (RPi-RP2) that the the RP2040 exposes from the ROM bootloader when BOOTSEL mode is entered is a "fake" volume. There is no mutable storage backing the volume. Instead the UF2 file sectors are recognized and processed as they are written. Other writes to the volume will succeed, but reading back the data will give the wrong contents.

**NOTE: This is a common mechanism employed by embedded devices which support firmware updates over USB**

Something has clearly changed in macOS Ventura; either

1) An additional re-read of the metadata after write (possibly incidental due to other changes)
2) An upgrading of errors based on reading incorrect data.
3) Perhaps some cache change whereby the data was always read back, but just happened to generally be in the disk cache as it is small and just-written

kilograham, 2022-10-26:
In any case, i would suggest the new behavior of macOS Ventura is incorrect; if you drag a file onto a USB drive, the intention is to copy the file, and that should succeed (if there is no disk failuer) irresepective of other metadata

jaustin, 2022-10-27:
We're tracking a very similar (likely the same...) issue with DAPLink, which is used for mbed and micro:bit devices amongst others.

liamfraser, 2022-10-27:
I took an image of the RPI-RP2 drive from a Pico using dd, and then wrote it to a normal flash drive. The contents after dragging blink.uf2 are:

```
liam@Liams-Air RPI-RP2 % find .
.
./INDEX.HTM
./INFO_UF2.TXT
./.fseventsd
./.fseventsd/fseventsd-uuid
./blink.uf2
./._blink.uf2
```

```
liam@Liams-Air RPI-RP2 % cat .fseventsd/fseventsd-uuid
4B657E01-DB98-4A7E-B389-CA0F1EA8821D%
liam@Liams-Air RPI-RP2 % cat ._blink.uf2
Mac OS X        	2��ATT�xThis resource fork intentionally left blank   ��
```

I don't know how this compares to a previous version. Perhaps I can spin up a VM of Monterey to compare.

aallan, 2022-10-27:
> Perhaps I can spin up a VM of Monterey to compare.

If you're in the office tomorrow you can grab someone (any of) the people from the Comms team that might be in, everyone there has a Mac, and none of them will be running Ventura at this point.

microbit-carlos, 2022-10-28:
In case it helps, I had issues creating a VM with a VMWare Fusion from Monterey on an x86 Macbook. I tried creating one from the Monterey recovery partition and upgrading, but it constantly crashed and I was completely unable to enter Recovery Mode (with any method I found online) to disable System Integrity Protection, which is useful for using additional utilities like ptrace/dtrace/wireshark.

On the other hand on a different x86 macbook already upgraded to Ventura, I was able to easily create a working VM from the recovery partition and disable SIP. I think Parallels then lets you easily transfer the VM to a different system, so that might be a useful way to upgrade only one computer and then use it to generate VMs that could be used from computers using  Monterey or older.

aallan, 2022-10-29:
> None of these are long term fixes and I fully recognize the backward compatibility problems of changing your SD cards and/or your firmware to work…

At least for RP2040, the firmware is on ROM on the silicon, we literally can't change it.

wolfpaulus, 2022-10-30:
Try the ditto command instead of cp.. maybe like so:
ditto --norsrc --noextattr --noacl ~/Desktop/demo.py /Volumes/RPI-RP2/

Cipulot, 2022-10-30:
I just tried to verify if the drag still produces the error. On the new beta, 13.1 (22C5033e) persists.

aallan, 2022-10-31:
Just to keep tabs on them, other folks that have Github issues open:

* micropython/micropython#9794
* ARMmbed/DAPLink#982

dhalbert, 2022-10-31:
We've published this on the Adafruit blog, which also goes to Twitter. We can revise the post if needed:
https://blog.adafruit.com/2022/10/31/uploading-uf2-files-with-macos-13-0-ventura-apple-microbit_edu-raspberry_pi-circuitpython/

aallan, 2022-11-01:
We've also now written this up as a blog post, https://www.raspberrypi.com/news/the-ventura-problem/.

aallan, 2022-11-01:
I've hacked together a quick macOS Shortcut. If you add it as a Quick Action — you'll need to give it permission to run shell scripts — then you can just right-click on your UF2 and select "Quick Actions > Copy to RPI-RP2" and your UF2 will be copied to your Pico. It's not as nice as drag-and-drop, but it's better than opening a Terminal window I guess? 

https://www.icloud.com/shortcuts/989b90c87fec4c11965dd517685f87cd

micheld93, 2022-11-07:
I gave feedback to Apple today. The number they gave me was FB11762117. I used your FB from the blog post to link the issue to RasPi.

chrisandrew-dev, 2022-11-13:
> Just to confirm for folks that aren't directly involved,
> 
> ```
> cp -X blink.uf2 /Volumes/RPI-RP2/
> ```
> 
> works as expected, copying the UF2 file to the virtual volume.

Hi there, this no longer works for me after updating to 13.0.1 (hoping it would resolve the drag/drop issue). For a complete beginner, it's pretty frustrating to say the least.

aallan, 2022-11-13:
It works for me on 13.0.1. Try,

```
rsync blink.uf2 /Volumes/RPI-RP2
```

instead?

Just to be sure, your Terminal window is in the same directory as the UF2 binary is in?

aallan, 2022-11-15:
According to comments on ARMmbed/DAPLink#982, `cp -X` doesn't work for firmware files >1MB in size. However `rsync` does work. I'd therefore recommend people look to use `rsync` rather than `cp` until there is a proper fix.

aallan, 2022-11-19:
> Hi there, this no longer works for me after updating to 13.0.1 (hoping it would resolve the drag/drop issue). For a complete beginner, it's pretty frustrating to say the least.

Confirmed that `cp -X` works as expected on 13.0.1. Sorry, but you must be experiencing some other problem that isn't related.

chrisandrew-dev, 2022-11-20:
> Confirmed that `cp -X` works as expected on 13.0.1. Sorry, but you must be experiencing some other problem that isn't related.

Thank you for checking, @aallan. I've been able to transfer a blink.UF2 file downloaded from the official website using the above method. Unfortunately, I'm still unable to detect and connect to the board (Arduino IDE), view or manage whatever files may be stored on there. I will keep troubleshooting...

aallan, 2022-11-21:
> Thank you for checking, @aallan. I've been able to transfer a blink.UF2 file downloaded from the official website using the above method. Unfortunately, I'm still unable to detect and connect to the [...]

Just to show it working! 

https://user-images.githubusercontent.com/53625/203112367-31ddaaca-8a55-44af-824a-4c461dd1e54d.mov

predmijat, 2022-11-28:
macOS 13.0.1, nothing works.
drag & drop 'error 100093'.
cp -X 'disk not ejected properly', same thing with rsync.

aallan, 2022-11-28:
I'm running 13.0.1 and both "cp -X" and "rsync" on the command line work to copy the file to the "fake" volume. As noted [in  he blog post](https://www.raspberrypi.com/news/the-ventura-problem/#comment-1588512) you should expect error messages, but you can ignore them, the copy completes correctly. 

You could try the short cut action if the messages bother you however?

> I've hacked together a quick macOS Shortcut. If you add it as a Quick Action — you'll need to give it permission to run shell scripts — then you can just right-click on your UF2 and select "Quick [...]

stuartziane, 2022-12-01:
Hey there, I built a macOS app that makes it really easy to upload UF2 files to the Pico (and any other RP2040 board technically).  You can get it for free here: [https://github.com/stuartziane/Pico-UF2-Uploader-Mac]

You'll still get the system notification about the Pico being ejected, but I decided to leave it like that, just because it's closer to how things normally happen.  If it's annoying, I'll sort that out.

Please let me know what you think of the app!

Cheers,

Stuart

stuartziane, 2022-12-02:
Sorry if you had trouble earlier trying to download the app - For some reason, the app itself was uploaded as a folder, rather than a .app.  It's fixed now.

aallan, 2022-12-07:
According to [a comment](https://github.com/ARMmbed/DAPLink/issues/982#issuecomment-1339852504) on the ARMmbed/DAPLink#982 issue on this, "I've tested macOS Ventura 13.1 beta.4 and it looks like it resolves the issue."

I've not yet had a chance to check this myself, or heard anything directly back from Apple, but if anyone that has access to macOS beta releases wants to try out the latest beta and report back?

kilograham, 2022-12-14:
This is fixed on macOS Ventura 13.1

closed 2022-12-14 by kilograham

LettError, 2023-01-13:
* Macos Ventura Version 13.1 (22C65)
* rp2-pico-w-20230113-unstable-v1.19.1-790-gedc92d18d.uf2

Drag and drop in Finder starts promising, with a grayed out name in the Pico window and a small progress icon. 

[image]

Then the RPI-RP2 unmounts and does not reappear. I need to disconnect and reconnect (with the button down) the uf2 file has not copied. 

Using rsync, cp -X also result in unmounting and disappearing and file not copying.

JamesH65, 2023-01-13:
The unmount is the correct behaviour, at that point the Pico should have rebooted and run your software. Rebooting with the button pressed makes it go back in to boot mode, the the UF2 does NOT appear in the file list as its a virtual file system only intended to support the copying of files. Are you sure your program is not running straight after the copy?

JamesH65, 2023-01-13:
If your code outputs anything on serial then monitor that, or turn on the onboard LED when you code starts running. I tend to flash the onboard if I need to check things are starting up.

carlosperate, 2023-01-13:
Based on the file filename (v1.19.1) it looks like you might be loading a MicroPython UF2 file. In that case, and if you haven't somehow attached your own code to the build, then it probably doesn't do anything "visible" when running and there is no MSD drive to load.
The easiest way to check it's running correctly is to connect to the serial port and it should drop you into the MicroPython REPL.

dhalbert, 2023-01-13:
> I'm... not sure. Maybe my expectation is wrong.. (for instance Adafruit Metro Express Circuit py mounts as a device _and_ runs the code.) Is there any way to tell the uf2 is installing properly?

CircuitPython presents the CIRCUITPY drive. MicroPython does _not_ present a drive, but serial access to the REPL should be available. You should see a `/dev/cu.something` device appear that you can connect to with a terminal program, Mu, etc.

LettError, 2023-01-13:
Expectations adjusted :) With https://thonny.org 4.01 for Mac and the settings that go with it, I have a blinking led. Thank you for the helpful comments!