raspberrypi/pico-sdk #1083: tinyusb tud_resume_cb() is not called and tud_suspended() returns wrong status

mrdudz, 2022-10-28:
My code is based on the hid_multiple_interface example. i have added an endpoint for "acpi powerbutton", which i use to put the host into sleep mode. Everything works as expected, except when my code initiates a remote wake up by calling tud_remote_wakeup() - in that case the host wakes up as expected, but for some reason the tud_resume_cb() is not called and tud_suspended() returns true (which then prevents any further USB transfers). If the wake up is initiated by the host, everything works fine (which makes me think my code is ok).

The behaviour changed after i did "git pull" in my pico-sdk directory - before the update using tud_remote_wakeup() would actually not work correctly, and i worked around this by using the acpi powerbutton "wake up" message. Which by itself is odd :) However, the  tud_suspended() flag did work correctly. Something must have been changed since then which made tud_remote_wakeup() work - but not correctly updating the internal flags to make tud_suspended() work.

mrdudz, 2022-10-29:
I am now doing this to be sure its not my code:

```
            tud_remote_wakeup();
            {
                uint64_t end = time_us_64() + (5000 * 1000);
                do {
                    tud_task();
                    if (time_us_64() > end) {
                        LOG(("WARNING: tud_suspended() still true after 5 seconds\n"));
                        break;
                    }
                } while (tud_suspended());
```

... it still runs into the timeout (when the host is clearly ready for some tme)

hathach, 2022-10-29:
what is the git hash of tinyusb you are using. There is a fix related to this https://github.com/hathach/tinyusb/pull/1542

mrdudz, 2022-10-30:
I have checked out the pico-sdk and tinyusb as explained in the "getting started" document, and only ever updated by doing "git pull", as updating individual sub modules is not recommended. That probably means tinyusb is at https://github.com/hathach/tinyusb/tree/4bfab30c02279a0530e1a56f4a7c539f2d35a293 - which does not contain that fix.

mrdudz, 2022-11-10:
Oh dear. I started over with a fresh clean pico-sdk clone and now it is working (and the linked fix seems to be there). No idea what was happening. Isnt "git pull" in the root of the sdk the correct way to update it? (including tinyusb?)

JamesH65, 2022-11-10:
> Oh dear. I started over with a fresh clean pico-sdk clone and now it is working (and the linked fix seems to be there). No idea what was happening. Isnt "git pull" in the root of the sdk the correct way to update it? (including tinyusb?)

See section 2.3 in this guide for how to update the SDK. https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

kilograham, 2023-05-26:
closing as this was likely fixed in 1.5.0 with TinyUSB upgrade

closed 2023-05-26 by kilograham